arduino/Arduino #9198: Can no longer build on MacOS following the build instructions (requires homebrew's JDK8, which no longer is available)

plocher, 2019-09-05:
> Mac OS X
> On Mac OS X (using Homebrew, first brew update), install:
> 
> Apple's Developer Tools
> git (brew install git)
> JDK 8

Using MacOS Mojave 10.14.6

$ uname -a
Darwin staging.local 18.7.0 Darwin Kernel Version 18.7.0: Thu Jun 20 18:42:21 PDT 2019; root:xnu-4903.270.47~4/RELEASE_X86_64 x86_64

$ brew cask install java8
==> Tapping homebrew/cask
Cloning into '/usr/local/Homebrew/Library/Taps/homebrew/homebrew-cask'...
remote: Enumerating objects: 3635, done.
remote: Counting objects: 100% (3635/3635), done.
remote: Compressing objects: 100% (3625/3625), done.
remote: Total 3635 (delta 24), reused 619 (delta 8), pack-reused 0
Receiving objects: 100% (3635/3635), 1.21 MiB | 7.41 MiB/s, done.
Resolving deltas: 100% (24/24), done.
Tapped 1 command and 3525 casks (3,640 files, 3.9MB).
Error: Cask 'java8' is unavailable: No Cask with this name exists.

plocher, 2019-09-05:
Workaround: Install openJDK8 from an alternate source, for example:
https://adoptopenjdk.net/releases.html#x64_mac

per1234, 2020-09-28:
Thanks so much for reporting this @plocher!

I believe the instructions were fixed by:
https://github.com/arduino/Arduino/wiki/Building-Arduino/_compare/6adfbc6f0609126bc437288d0423309aab862df0...dfdad803a9119e10c69698c61cf3338881d93c7f

They work for me anyway. So I'm going to close this. If I was mistaken and the problem remains, please comment here and I'll investigate.

closed 2020-09-28 by per1234