arduino/Arduino #9199: New library

offcircuit, 2019-09-06:
Hi,

Can I add a  library?

https://github.com/offcircuit/GPS

Thanks,
Helder Rodrigues

per1234, 2019-09-06:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

offcircuit, 2019-09-06:
Hi again,

Done,

:)
Helder Rodrigues

offcircuit, 2019-09-08:
... now ...

per1234, 2019-09-13:
@offcircuit I notice you closed this issue. Did you decide that you didn't want your library added to the Arduino Library Manager index after all?

offcircuit, 2019-09-14:
hi again,

I will take abetter look at this class

Thanks,
Helder Rodrigues

offcircuit, 2019-10-01:
Hi again,

I made some changes on the lib.

Now it is good to publish,
https://github.com/offcircuit/GPS

Thanks,
Helder Rodrigues

cmaglie, 2019-10-03:
done

closed 2019-10-03 by cmaglie