arduino/Arduino #9202: Win 10 Arduino Exit 1 Error Permission Denied | Need Urgent Help | Code not Compiling

hurtlocker7, 2019-09-08:
Arduino: 1.8.9 (Windows 10), Board: "Arduino/Genuino Uno"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Vaibhav Dixit\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10809 -build-path C:\Users\VAIBHA~1\AppData\Local\Temp\arduino_build_529793 -warnings=all -build-cache C:\Users\VAIBHA~1\AppData\Local\Temp\arduino_cache_158768 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Vaibhav Dixit\Documents\Arduino\libraries -fqbn=arduino:avr:uno -ide-version=10809 -build-path C:\Users\VAIBHA~1\AppData\Local\Temp\arduino_build_529793 -warnings=all -build-cache C:\Users\VAIBHA~1\AppData\Local\Temp\arduino_cache_158768 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Program Files (x86)\Arduino\examples\01.Basics\Blink\Blink.ino
Using board 'uno' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Using core 'arduino' from platform in folder: C:\Program Files (x86)\Arduino\hardware\arduino\avr
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\VAIBHA~1\\AppData\\Local\\Temp\\arduino_build_529793\\sketch\\Blink.ino.cpp" -o nul
Generating function prototypes...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10809 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\standard" "C:\\Users\\VAIBHA~1\\AppData\\Local\\Temp\\arduino_build_529793\\sketch\\Blink.ino.cpp" -o "C:\\Users\\VAIBHA~1\\AppData\\Local\\Temp\\arduino_build_529793\\preproc\\ctags_target_for_gcc_minus_e.cpp"
cc1plus.exe: fatal error: opening output file C:\Users\VAIBHA~1\AppData\Local\Temp\arduino_build_529793\preproc\ctags_target_for_gcc_minus_e.cpp: Permission denied

compilation terminated.

exit status 1
Error compiling for board Arduino/Genuino Uno.Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.

per1234, 2019-09-08:
This sort of error could be caused by your antivirus software. Try **TEMPORARILY** disabling your antivirus for a single compilation to see if the problem goes away, then turn the antivirus back on. If the problem doesn't occur with the antivirus off you will need to adjust the settings of your antivirus to whitelist the appropriate file, folder, or process so it doesn't interfere with compilation.

Please let us know if that solves the error.

hurtlocker7, 2019-09-09:
> This sort of error could be caused by your antivirus software. Try **TEMPORARILY** disabling your antivirus for a single compilation to see if the problem goes away, then turn the antivirus back on. [...]

Problem is still same no progress after all the changes and granted permissions to even temp folders. But i tired today on many win 10 systems. Its working fine. But with my asus its not.

shreyas1212, 2020-07-25:
Ok, Same issue with me 

I have just cleaned my Temp folder and working fine, Please try.

(type %temp% in RUN windows and delete all from temp folder