arduino/Arduino #9203: [Library Manager] Remove a library and add a new one.

jaean123, 2019-09-09:
I would like to remove this library from the library manager.
https://www.arduinolibraries.info/libraries/snappy-xo-precise-movement-library

And add this new library. Although the name is the same, it's a new but similar library. 
https://github.com/purwar2016/PreciseMovement-library

Thank you for your time.

cmaglie, 2019-09-14:
I've added the library from @purwar2016 repository,

BTW I don't understand which library you want to remove. The library added is named `PreMo - Virtual Path Following` and there are no other libraries with the same name. Also the link you posted is broken...

jaean123, 2019-09-14:
@cmaglie. It looks like it's resolved now. Thanks.

closed 2019-09-14 by jaean123