arduino/Arduino #9208: [Library Manager] New Library: MicroTone

Master811129, 2019-09-10:
Hi there,
could you please add the following library at the Library Manager:

https://github.com/Master811129/MicroTone

thanks

per1234, 2019-09-10:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Master811129, 2019-09-10:
> You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file. [...]

Uuummm
I think i got that. can you re-check it for me?

cmaglie, 2019-09-14:
done

closed 2019-09-14 by cmaglie

Master811129, 2019-09-14:
> done

thank you sir but how can i put an update on it

per1234, 2019-09-14:
From the Arduino Library Manager FAQ:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-publish-a-new-release-once-my-library-is-in-the-list
>How can I publish a new release once my library is in the list?

Ensure you've changed `version` in your library.properties. Then tag your library once more and push the new tag (or create a release if your web hosting offers a way to do it, for example with GitHub "releases"). Our indexer checks for new releases every hour and will eventually fetch and publish your new release.

njh, 2024-01-02:
@Master811129 it looks like the MicroTone library has now been deleted - do you have any information about why?