arduino/Arduino #9215: ++ operator used for array index ( array[index++] )

ionicat, 2019-09-11:
After executing the following code:
	int i=0;
	char s1[20];
	char s2[20];
	strcpy(s1, "test1");
	strcpy(s2, "xxxxxx");
	while(i<4)
		s2[++i] = s1[i];
	s2[i++] = '2';
	s2[i] = 0;
on Visual Studio (Windows) 
s1 = "test1"
s2 = "test2"
as expected
however, on Arduino
s1 = "test1"
s2 = "est12"
Arduino acts like 
	while(i<4)
		s2[i++] = s1[i];
is equivalent with
	while(i<4){
		s2[i] = s1[i+1];
		i++;
	}
instead of
	while(i<4){
		s2[i] = s1[i];
		i++;
	}

This is just a simplified code to show the issue discovered in a more complex situation, I know that something like
	int i=-1;
	while(i++<4)
		s2[i] = s1[i];
would have been fine

sandeepmistry, 2019-09-16:
Hi @ionicat,

What board are you using?

ionicat, 2019-09-17:
Hi,
It is an Arduino Pro Mini Board