arduino/Arduino #9229: Updating arduino removes boards.

a-c-sreedhar-reddy, 2019-09-16:
When I updated the Arduino all the boards are gone and I need to install them all. How can we update the Arduino with the same boards?

per1234, 2019-09-16:
How did you update?

Which version did you update from?

Which version did you update to?

How did you install the boards?

a-c-sreedhar-reddy, 2019-09-16:
I installed the folder for linux and ran install.sh script.
I could not remember from which version I upgraded from.
I updated to the latest.
I installed esp32 boards using https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/debian_ubuntu.md

per1234, 2019-09-16:
What is the folder location of your previous Arduino IDE installation?

Were you running the Arduino IDE in portable mode (https://www.arduino.cc/en/Guide/PortableIDE)?

What is the location of your sketchbook folder (File > Preferences > Sketchbook location)?

per1234, 2020-08-04:
Closing due to lack of response.

@a-c-sreedhar-reddy if you'll provide the requested information I'll be happy to re-open this issue.

closed 2020-08-04 by per1234