arduino/Arduino #9231: [Library Manager] Please add Otto DIY libraries

cparrapa, 2019-09-16:
Hi Arduino team

Could you please add the following libraries at the Library Manager:

https://github.com/OttoDIY/Libraries

https://github.com/OttoDIY/Libraries/releases/tag/0.0.1

Thanks in advance

cmaglie, 2019-10-03:
> https://github.com/OttoDIY/Libraries/releases/tag/0.0.1

@cparrapa this is not going to work, you must put each library in a different repository.

We also require that each library be located in the root of the repository. In your case the libraries are in different subfolders of the main repository, which is not allowed (but this is a consequence of having all the libraries together).

After moving the libraries, remember to either create a new tag/release. Once that's done, comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

cparrapa, 2019-10-06:
Hi @cmaglie 
Thanks for the review, we will see and work on how to unify or separate the libraries then with a better naming to be easy to find, thanks

cmaglie, 2020-02-17:
I'm closing this for now, please open a new request when the library is ready.

closed 2020-02-17 by cmaglie

cparrapa, 2020-02-17:
Hi @cmaglie 
yes is fine we did not manage to combine them into one it will require separate libraries and repos it seems

Thanks