arduino/Arduino #9234: Issue with Serial Monitor in release 1.8.10

gpm-102, 2019-09-17:
I have just updated from 1.8.9 to 1.8.10, and am having a problem with the serial monitor.
I am working on a system using motor shields and motors. To assist with debugging I have serial.print statements to output to the monitor. When things go wrong I either need to remove the motor power, or disconnect the usb to stop the hardware getting damaged. 
When I do this in 1.8.10 the displayed information disappears, meaning I can't track what has happened. This did not happen in 1.8.9.
Using Arduino Mega 2560
on Win 10 computer

facchinm, 2019-09-18:
Hi @gpm-102 ,
on 1.8.10 the serial monitor freezes when it recognizes that a board has been disconnected. Then, if the board comes back online, it reopens it. This avoids a lot of bugs related to the serial monitor hogging the cpu while trying to read a detached serial.
 
Your use case is a bit borderline (I must admit I didn't think about it). Probably a freeze without graying out the window would work in your case, so you could still see the last messages.

@cmaglie any thoughts on that?

cmaglie, 2019-09-18:
> Probably a freeze without graying out the window would work in your case, so you could still see the last messages

Looks like a good compromise :-)

(if we ever get to grey-out keeping the messages visibile, even better)

gpm-102, 2019-09-18:
I can see your logic for making this change.
A freeze without greying out probably wouldn't help much, as I need to scroll back through the history.
I will either have to find an alternative way of working, or go back to 1.8.9.
It will be interesting to see if others meet the same problem.

gpm-102, 2019-09-18:
I have work round which may help others.
I have a hardware start switch, which I read and wait at until it is pressed. I have arranged it so that all the motors are set up and set to zero speed before this point.
This means that when there is a problem I can press the Reset button, and everything stops and waits at the Start switch.
Fortunately Reset does not kill the Monitor screen

PaulStoffregen, 2019-11-06:
> (if we ever get to grey-out keeping the messages visibile, even better)

This is what Teensy's not-yet-pluggable serial monitor does.  When the device disconnects, the window is disabled but remains visible.

Even that wasn't great, because people wanted to be able to copy text to the clipboard.  Recently I changed it so the TextArea part is kept enabled, but turned gray by setting its foreground & background colors.

I've been intending to send a huge pull request with all this stuff, including the speedup that allows the serial monitor to handle 10 Mbyte/sec speed.  Right now I'm working on a critical project, then traveling later this month.  Maybe in December, if there's interest?

PaulStoffregen, 2019-11-06:
If anyone wants that code before I get around to cleaning it up and making it more "pluggable", it's all published here:

https://github.com/PaulStoffregen/Arduino-1.8.10-Teensyduino/blob/master/app/src/processing/app/TeensyPipeMonitor.java

The part about disabling but leaving the window visible and keeping the textarea able to select & copy text is in the onEnableWindow() function, on lines 190-218.

I also wrote this blog article about the performance improvement work.

https://www.pjrc.com/improving-arduino-serial-monitor-performance/

Near the end is a section about thoughts on contributing this code, which has links to the specific source files.  I also wrote some lengthy comments in FifoDocument.java.  Sadly, I just haven't had time to clean this up and turn it into a more generic "pluggable" version, as was done for pluggable discovery.  So many extremely urgent things always seem to drain away all the available dev time when you run a small business.  I will eventually get around to it... but if anyone wants the code sooner, it's all published as open source for anyone to take & use & improve.

PR #9644 (2020-02-12): Do not clear serial monitor if the port is disabled for upload
Fix #9234 

Used the patch provided by @PaulStoffregen
commit 7ae377d3 (2020-02-12): Do not clear serial monitor when disabled for upload (@PaulStoffregen)

Fix #9234
closed 2020-02-12 by cmaglie