arduino/Arduino #9242: Error plz help me

bhavindholu87, 2019-09-20:
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x7855fb]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.ResolveLibrary(0x118d45a0, 0x1195ca41, 0x5, 0x1195ca41)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/resolve_library.go:64 +0x14b
github.com/arduino/arduino-cli/legacy/builder.findIncludesUntilDone(0x118d45a0, 0x118f4f00, 0x88ed00, 0x119a2690, 0x11843fa8, 0x11af3600, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:358 +0x3b6
github.com/arduino/arduino-cli/legacy/builder.(*ContainerFindIncludes).Run(0xdebc68, 0x118d45a0, 0xdebc68, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:152 +0x3e5
github.com/arduino/arduino-cli/legacy/builder.runCommands(0x118d45a0, 0x11865e1c, 0x20, 0x20, 0x409a01, 0x3ffff, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xbd
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0x11871f40, 0x118d45a0, 0xdebbb4, 0x11842c70)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:117 +0xb9c
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:226
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:398 +0x6d2
C:\Program Files (x86)\Arduino\arduino-builder returned 2
Error compiling for board Arduino/Genuino Uno.

cmaglie, 2019-09-20:
could you post some more information? Like the board you're using, a copy of the sketch, the version of the IDE....

bhavindholu87, 2019-09-20:
Version - 1.8.10
I'm using the arduino UNO
I'm work on i2c lcd display 16x2

cmaglie, 2019-09-20:
- which operating system?
- set "Verbose compile" from the preferences and copy&paste the full output.
- copy&paste also the sketch

bhavindholu87, 2019-09-20:
this code in coming error:
#include <LiquidCrystal_I2C.h>

#include "Wire.h" // For I2C
#include "LCD.h" // For LCD
#include "LiquidCrystal_I2C.h" // Added library*
//Set the pins on the I2C chip used for LCD connections
//ADDR,EN,R/W,RS,D4,D5,D6,D7
LiquidCrystal_I2C lcd(0x27,2,1,0,4,5,6,7); // 0x27 is the default I2C bus address of the backpack-see article
void setup()
{
   // Set off LCD module
   lcd.begin (16,2); // 16 x 2 LCD module
   lcd.setBacklightPin(3,POSITIVE); // BL, BL_POL
   lcd.setBacklight(HIGH);
  
}
void loop()
{
   lcd.setCursor(5,0);
   lcd.print("Marwadi");}

bhavindholu87, 2019-09-20:
error is 

panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x7855fb]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.ResolveLibrary(0x11cba5a0, 0x11d513d6, 0x5, 0x11d513d6)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/resolve_library.go:64 +0x14b
github.com/arduino/arduino-cli/legacy/builder.findIncludesUntilDone(0x11cba5a0, 0x11cd8680, 0x88ed00, 0x11d7e690, 0x11ec53e0, 0x11ecf400, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:358 +0x3b6
github.com/arduino/arduino-cli/legacy/builder.(*ContainerFindIncludes).Run(0xdebc68, 0x11cba5a0, 0xdebc68, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:152 +0x3e5
github.com/arduino/arduino-cli/legacy/builder.runCommands(0x11cba5a0, 0x11c55e1c, 0x20, 0x20, 0x409a01, 0x7fff, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xbd
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0x11c61f40, 0x11cba5a0, 0xdebbb4, 0x11c06c80)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:117 +0xb9c
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:226
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:398 +0x6d2
C:\Program Files (x86)\Arduino\arduino-builder returned 2
Error compiling for board Arduino/Genuino Uno.

cmaglie, 2019-09-20:
There is something going on in your particular installation but, to understand what the problem is, I need you to enable the "verbose compile" by flagging the check box in the preferences window as in the following image:

[image]

After doing so, please, verify again the sketch and copy the output here. The output should be much longer and detailed.

cmaglie, 2019-11-07:
@bhavindholu87 did you solve the issue? may you give the requested information?

cmaglie, 2019-11-08:
Fixed by https://github.com/arduino/arduino-cli/pull/477, the patch will be release with IDE 1.8.11

closed 2019-11-08 by cmaglie

emilijaboba, 2020-01-01:
Hi, I have the same problem, but with Nextion display. Here's the error message:

Arduino: 1.8.10 (Windows 10), Board: "Arduino Due (Programming Port)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Bazi\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Bazi\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Bazi\Documents\Arduino\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X2341_0X003D -ide-version=10810 -build-path C:\Users\Bazi\AppData\Local\Temp\arduino_build_829202 -warnings=all -build-cache C:\Users\Bazi\AppData\Local\Temp\arduino_cache_19294 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\Bazi\Desktop\First_try_with_NEXTION_display_v.1\Nextion_Tutorial2\Nextion_Tutorial2.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Bazi\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Bazi\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Bazi\Documents\Arduino\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X2341_0X003D -ide-version=10810 -build-path C:\Users\Bazi\AppData\Local\Temp\arduino_build_829202 -warnings=all -build-cache C:\Users\Bazi\AppData\Local\Temp\arduino_cache_19294 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\Bazi\Desktop\First_try_with_NEXTION_display_v.1\Nextion_Tutorial2\Nextion_Tutorial2.ino
Using board 'arduino_due_x_dbg' from platform in folder: C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12
Using core 'arduino' from platform in folder: C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12
Detecting libraries used...
"C:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10810 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "C:\\Users\\Bazi\\AppData\\Local\\Temp\\arduino_build_829202\\sketch\\Nextion_Tutorial2.ino.cpp" -o nul
Alternatives for Nextion.h: [Nextion Nextion ITEADLIB_Arduino_Nextion-master NeoNextion@2.2.0]
ResolveLibrary(Nextion.h)
  -> candidates: [Nextion Nextion ITEADLIB_Arduino_Nextion-master NeoNextion@2.2.0]
"C:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10810 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\core [... cut ...]
    Serial.print(variable1);  // This is the value you want to send to that object and atribute mentioned before.
    Serial.write(0xff);  // We always have to send this three lines after each command sent to the nextion display.
    Serial.write(0xff);
    Serial.write(0xff);
  
  }

  // We are going to check the list of touch events we enter previously to
  // know if any touch event just happened, and excecute the corresponding instructions:

  nexLoop(nex_listen_list);  // Check for any touch event

}  // End of loop

emilijaboba, 2020-01-01:
Hi, I have the same problem, but with Nextion display. Also I turned on verbile compiling. Here's the error message:

Arduino: 1.8.10 (Windows 10), Board: "Arduino Due (Programming Port)"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Bazi\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Bazi\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Bazi\Documents\Arduino\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X2341_0X003D -ide-version=10810 -build-path C:\Users\Bazi\AppData\Local\Temp\arduino_build_829202 -warnings=all -build-cache C:\Users\Bazi\AppData\Local\Temp\arduino_cache_19294 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\Bazi\Desktop\First_try_with_NEXTION_display_v.1\Nextion_Tutorial2\Nextion_Tutorial2.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\Bazi\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\Bazi\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\Bazi\Documents\Arduino\libraries -fqbn=arduino:sam:arduino_due_x_dbg -vid-pid=0X2341_0X003D -ide-version=10810 -build-path C:\Users\Bazi\AppData\Local\Temp\arduino_build_829202 -warnings=all -build-cache C:\Users\Bazi\AppData\Local\Temp\arduino_cache_19294 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.bossac.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.bossac-1.6.1-arduino.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\bossac\1.6.1-arduino -prefs=runtime.tools.arm-none-eabi-gcc.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -verbose C:\Users\Bazi\Desktop\First_try_with_NEXTION_display_v.1\Nextion_Tutorial2\Nextion_Tutorial2.ino
Using board 'arduino_due_x_dbg' from platform in folder: C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12
Using core 'arduino' from platform in folder: C:\Users\Bazi\AppData\Local\Arduino15\packages\arduino\hardware\sam\1.6.12
Detecting libraries used...
"C:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -c -g -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib -fno-threadsafe-statics --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -w -x c++ -E -CC -mcpu=cortex-m3 -mthumb -DF_CPU=84000000L -DARDUINO=10810 -DARDUINO_SAM_DUE -DARDUINO_ARCH_SAM -D__SAM3X8E__ -mthumb -DUSB_VID=0x2341 -DUSB_PID=0x003e -DUSBCON "-DUSB_MANUFACTURER=\"Arduino LLC\"" "-DUSB_PRODUCT=\"Arduino Due\"" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/libsam" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/CMSIS/Include/" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\system/CMSIS/Device/ATMEL/" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\cores\\arduino" "-IC:\\Users\\Bazi\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\sam\\1.6.12\\variants\\arduino_due_x" "C:\\Users\\Bazi\\AppData\\Local\\Temp\\arduino_build_829202\\sketch\\Nextion_Tutorial2.ino.cpp" -o nul
Alternatives for doxygen.h: [ITEADLIB_Arduino_Nextion-master Nextion Nextion]
ResolveLibrary(doxygen.h)
  -> candidates: [ITEADLIB_Arduino_Nextion-master Nextion Nextion]
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x7855fb]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.ResolveLibrary(0x11cc25a0, 0x11f2c0a5, 0x9, 0x11f2c0a5)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/resolve_library.go:64 +0x14b
github.com/arduino/arduino-cli/legacy/builder.findIncludesUntilDone(0x11cc25a0, 0x11eec0e0, 0x88ed00, 0x11da0690, 0x11c06758, 0x11d14000, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:358 +0x3b6
github.com/arduino/arduino-cli/legacy/builder.(*ContainerFindIncludes).Run(0xdebc68, 0x11cc25a0, 0xdebc68, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:152 +0x3e5
[... cut ...]
    Serial.print(variable1);  // This is the value you want to send to that object and atribute mentioned before.
    Serial.write(0xff);  // We always have to send this three lines after each command sent to the nextion display.
    Serial.write(0xff);
    Serial.write(0xff);
  
  }

  // We are going to check the list of touch events we enter previously to
  // know if any touch event just happened, and excecute the corresponding instructions:

  nexLoop(nex_listen_list);  // Check for any touch event

}  // End of loop

udinpedia, 2020-05-16:
@cmagile 
I followed your instruction about "verbose compile" and here is the error
BTW I had the same problem here, but I use Arduino Nano board

Arduino: 1.8.10 (Windows 10), Board: "Arduino Nano, ATmega328P"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\HP\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10810 -build-path C:\Users\HP\AppData\Local\Temp\arduino_build_206896 -warnings=none -build-cache C:\Users\HP\AppData\Local\Temp\arduino_cache_463400 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\HP\Downloads\Gauss_Meter_Arduino_Code_for_google_drive\Gauss_Meter_Arduino_Code_for_google_drive.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\HP\Documents\Arduino\libraries -fqbn=arduino:avr:nano:cpu=atmega328 -vid-pid=1A86_7523 -ide-version=10810 -build-path C:\Users\HP\AppData\Local\Temp\arduino_build_206896 -warnings=none -build-cache C:\Users\HP\AppData\Local\Temp\arduino_cache_463400 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=C:\Program Files (x86)\Arduino\hardware\tools\avr -verbose C:\Users\HP\Downloads\Gauss_Meter_Arduino_Code_for_google_drive\Gauss_Meter_Arduino_Code_for_google_drive.ino
Using board 'nano' from platform in folder: C:\Program
Using core 'arduino' from platform in folder: C:\Program
Detecting libraries used...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "C:\\Users\\HP\\AppData\\Local\\Temp\\arduino_build_206896\\sketch\\Gauss_Meter_Arduino_Code_for_google_drive.ino.cpp" -o nul
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_NANO -DARDUINO_ARCH_AVR "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\cores\\arduino" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\variants\\eightanaloginputs" "-IC:\\Program Files (x86)\\Arduino\\hardware\\arduino\\avr\\libraries\\Wire\\src" "C:\\Users\\HP\\AppData\\Local\\Temp\\arduino_build_206896\\sketch\\Gauss_Meter_Arduino_Code_for_google_drive.ino.cpp" -o nul
Alternatives for LCD.h: [fmalpartida-new-liquidcrystal-1c093ac3a691 NewLiquidCrystal_lib]
ResolveLibrary(LCD.h)
  -> candidates: [fmalpartida-new-liquidcrystal-1c093ac3a691 NewLiquidCrystal_lib]
panic: runtime error: invalid memory address or nil pointer dereference
[signal 0xc0000005 code=0x0 addr=0x0 pc=0x7855fb]

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder.ResolveLibrary(0x11cc05a0, 0x11ffc55f, 0x5, 0x11ffc55f)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/resolve_library.go:64 +0x14b
github.com/arduino/arduino-cli/legacy/builder.findIncludesUntilDone(0x11cc05a0, 0x11ef3440, 0x88ed00, 0x11d86690, 0x12031128, 0x11ed3700, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:358 +0x3b6
github.com/arduino/arduino-cli/legacy/builder.(*ContainerFindIncludes).Run(0xdebc68, 0x11cc05a0, 0xdebc68, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_find_includes.go:152 +0x3e5
[... cut ...]
Invalid library found in C:\Users\HP\Documents\Arduino\libraries\gauss_meter_II: no headers files (.h) found in C:\Users\HP\Documents\Arduino\libraries\gauss_meter_II
Invalid library found in C:\Users\HP\Documents\Arduino\libraries\LCD: no headers files (.h) found in C:\Users\HP\Documents\Arduino\libraries\LCD
Invalid library found in C:\Users\HP\Documents\Arduino\libraries\gauss_meter_II: no headers files (.h) found in C:\Users\HP\Documents\Arduino\libraries\gauss_meter_II
Invalid library found in C:\Users\HP\Documents\Arduino\libraries\LCD: no headers files (.h) found in C:\Users\HP\Documents\Arduino\libraries\LCD