arduino/Arduino #9245: Lillypad arduino.. Board rate

gowdetilikhith, 2019-09-21:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
How to change a bord rate to 57600

per1234, 2019-09-22:
This issue tracker is only to be used to report bugs or feature requests. This topic is more appropriate for the Arduino Forum. I'm sure we'll be able to help you with your problem over there.

Please do this:
1. Read http://forum.arduino.cc/index.php?topic=148850
2. If you haven't already done so, create a new thread in the appropriate section of the forum (http://forum.arduino.cc/), following all the rules. Be sure to provide a detailed explanation of what you're trying to accomplish.

closed 2019-09-22 by per1234

gowdetilikhith, 2019-09-22:
Thank you ... For your suggesion