arduino/Arduino #9250: [Library Manager] Fix SparkFun LIS3DH library

nseidle, 2019-09-22:
Our library is not showing up in the library manager: https://github.com/sparkfun/SparkFun_LIS3DH_Arduino_Library

Here is the [original issue request](https://github.com/arduino/Arduino/issues/5800). I suspect it is because we messed up and changed the [original lib properties](https://github.com/sparkfun/SparkFun_LIS3DH_Arduino_Library/blob/bd75851c64df3d3bda6c037756ac6492fc3c8d91/library.properties) from:

name=SparkFun LIS3DH Breakout

to

name=SparkFun LIS3DH Arduino Library

Perhaps it's because the [original request](https://github.com/sparkfun/SparkFun_LIS3DH_Arduino_Library/tree/V_1.0.0) included the tree version. Not sure. Let me know how to fix. Perhaps the easiest is to delete and re-add.

cmaglie, 2019-10-03:
I've change the library name to `SparkFun LIS3DH Arduino Library` and removed `SparkFun LIS3DH Breakout`

closed 2019-10-03 by cmaglie

bboyho, 2019-10-03:
Thanks! The library manager can now see the different library versions. =)