arduino/Arduino #9261: Arduino Drivers expired

EricSP2, 2019-09-27:
Hello,

Would it be possible to update the signature of the Arduino drivers?

Some of them are out of date since 2016/2017...

With out of date drivers, it makes it impossible to silently deploy the Arduino drivers.

Please update it if possible.

cmaglie, 2019-09-27:
The signature should be "timestamped": it will not expire, even if the certificate used to sign the drivers expires in the meantime.

The timestamp has been added exactly for this purpose, so it's not needed to update the drivers every 2/3 years.

Are you getting errors about invalid signature?

EricSP2, 2019-09-27:
yes...

At the end of the day this is my fix:

```
REM Install the SmartPods Arduino drivers.
CertUtil -AddStore TrustedPublisher "Certs\Arduino srl.cer"
CertUtil -AddStore TrustedPublisher "Certs\Arduino LLC.cer"
PUSHD .\Drivers
dpinst-amd64.exe /LM /S /SE
POPD
DEVCON RESCAN
CertUtil -DelStore TrustedPublisher "Arduino srl"
CertUtil -DelStore TrustedPublisher "Arduino LLC"
```

EricSP2, 2019-09-27:
If I don't do the above, the publisher is untrusted for the driver.

I tried adding the Symantec sh256 cert, the arduino srl. arduino llc and the GlobalSign to the trusted root cert and no-go...

But adding the Arduino certs to the TrustedPublisher temporarily, that fixed it...

EricSP2, 2019-09-27:
Also if you look at the source of the signature in the trust chain, those certs in certain cases are also expired.

You may want to investigate, it's an issue with many sccm admins :/

EricSP2, 2019-09-27:
https://i.imgur.com/qM9ouba.jpg

EricSP2, 2019-09-27:
Please see the screen capture above.

cmaglie, 2019-09-27:
The screen you posted is the dialog for a *signed* driver, the **unsigned** is like this:

[image]

See also: https://www.davidegrayson.com/signing/

As I said, the drivers are "timestamped", there is no reason to re-sign them.
more on timestamp: https://security.stackexchange.com/a/85848

At this point I don't know if you can install drivers silently, it seems that at least the dialog you posted must be displayed.

feikname, 2019-09-28:
Only thing I see in the screenshot is Windows asking to confirm if you want to install drivers, which is a normal confirmation prompt when you don't already trust a publisher.

The solution for this is to [disable driver signature enforcement](https://windowsreport.com/would-you-like-to-install-this-device-software/) or do as you did.

The reason some drivers are able to install silently without this prompt is they get [verified by Microsoft](https://en.wikipedia.org/wiki/WHQL_Testing), which Arduino perhaps could apply to.

PaulStoffregen, 2019-09-29:
Is this driver installation necessary on Windows 10?  Microsoft bundles the needed INF for CDC ACM serial with Windows 10.

facchinm, 2019-09-30:
I have a PR ready to avoid installing any ACM driver on Win10  (to avoid issues like this https://github.com/arduino/ArduinoCore-samd/issues/423#issuecomment-505934306 ).
Unfortunately, there seem to be no clean way to uninstall them silently if they were already installed.

cmaglie, 2020-01-21:
@EricSP2 
I think that the solution you find is the only way to accomplish this, so I'm closing this issue. If you have other concerns feel free to comment further.

closed 2020-01-21 by cmaglie

EricSP2, 2020-01-21:
So at the end of the day, the Arduino is simply not trusted by Microsoft, so the only way to silently install the driver was to add the certificate temporarily during the installation process.

EricSP2, 2020-01-21:
> Is this driver installation necessary on Windows 10? Microsoft bundles the needed INF for CDC ACM serial with Windows 10.

The issue is that **SmartPods** client needs a very specific Arduino driverset in order to operate correctly.

You are correct that the built-in one works.

However with my above statement here was the issue:

IF I installed the SmartPods software silently and ignored the Arduino driver install and plugged in the smart desk, the software would finish the installation. However, the SmartPods client wouldn't natively recognize the desk hardware due to the driver STRING name it gave it using the built-in windows drivers. So essentially the client would deploy silently, but then every time I'd go to a desk, I'd have to manually define the arduino device on the COM port it assigned itself.

By pushing the real arduino drivers, the client software immediately recognized the desk and no manual intervents was needed.

So at the end of the day, yes the built-in driver works but it's an issue with the developers of smartpods for putting some odd parameter where the client software looks for a very specific driver title string.

So knowing those facts, I've brought it up with the developers and in the future they'll update the software to recognize the built-in driver name string.

marianvoicu2002, 2020-01-24:
Hello,

I use the following lines to install arduino uno drivers
PUSHD .\Drivers
CertUtil -AddStore TrustedPublisher "Certs\CertDrv1.cer"
CertUtil -AddStore TrustedPublisher "Certs\CertDrv2.cer"
CertUtil -AddStore TrustedPublisher "Certs\CertDrv3.cer"
CertUtil -AddStore TrustedPublisher "Certs\CertDrv4.cer"
dpinst-x86.exe /LM /S /SE
POPD
CertUtil -DelStore TrustedPublisher "Adafruit Industries"
CertUtil -DelStore TrustedPublisher "Arduino AG"
CertUtil -DelStore TrustedPublisher "Arduino srl"
CertUtil -DelStore TrustedPublisher "Arduino LLC"

It works for Windows7 X86, but doesn't work for WindowsXp x86.

On Xp works if i remove /S, and install with setup.

Can anyone help me?

TX,
Marian