arduino/Arduino #9266: CLI should look in sketchbook for sketch

slomobileAdmin, 2019-09-29:
According to [the man page](https://github.com/arduino/Arduino/blob/master/build/shared/manpage.adoc)
arduino [FILE.ino…​]
"A single .ino file should be given. If the sketch contains multiple .ino files, any one can be specified on the commandline, but the entire sketch will be compiled."

It says nothing about requiring an absolute path.  It is convenient and reasonable to assume, given just a sketch file name, arduino will look in the sketchbook for a folder with the same name as sketch minus .ino, then look for the sketch and use it.  It does not.

    slo@sloNano:~$ arduino --version
    Arduino: 1.8.10
    slo@sloNano:~$ arduino --get-pref sketchbook.path
    /home/slo/Arduino
    slo@sloNano:~$  --verify /home/slo/Arduino/Blink/Blink.ino
    Verifying...
    Sketch uses 12160 bytes (0%) of program storage space. Maximum is 2031616 bytes.
    Global variables use 18528 bytes (1%) of dynamic memory, leaving 1030048 bytes for local 
    variables. Maximum is 1048576 bytes.
    slo@sloNano:~$  --verify Blink.ino
    Verifying...
    java.io.IOException: No valid code files found
    	at processing.app.Sketch.listSketchFiles(Sketch.java:117)
	at processing.app.Sketch.<init>(Sketch.java:54)
	at processing.app.Base.<init>(Base.java:415)
	at processing.app.Base.main(Base.java:150)
    java.io.IOException: Stream closed
	at java.io.BufferedInputStream.getBufIfOpen(BufferedInputStream.java:170)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:336)
	at com.fasterxml.jackson.core.json.ByteSourceJsonBootstrapper.ensureLoaded(ByteSourceJsonBootstrapper.java:522)
	at com.fasterxml.jackson.core.json.ByteSourceJsonBootstrapper.detectEncoding(ByteSourceJsonBootstrapper.java:150)
	at com.fasterxml.jackson.core.json.ByteSourceJsonBootstrapper.constructParser(ByteSourceJsonBootstrapper.java:246)
	at com.fasterxml.jackson.core.JsonFactory._createParser(JsonFactory.java:1315)
	at com.fasterxml.jackson.core.JsonFactory.createParser(JsonFactory.java:820)
	at cc.arduino.packages.discoverers.PluggableDiscovery.run(PluggableDiscovery.java:80)
	at java.lang.Thread.run(Thread.java:748)

facchinm, 2019-09-30:
Probably related with https://github.com/arduino/arduino-builder/pull/342 . @cmaglie should we provide an hourly build with that builder?

cmaglie, 2019-09-30:
I think this is java-related, @slomobileAdmin which version of the IDE are you running?

slomobileAdmin, 2019-09-30:
The above was the Linux ARM 64bit version of the IDE running on a Jetson Nano with Ubuntu 18.04 with MATE desktop installed via apt-get, local terminal session.  Teensyduino 1.48 beta #1 is also installed and generating Java exceptions but I assumed they were unrelated to the above issue so I stripped them out of the above quoted terminal session for readability, perhaps I stripped too much.  Information regarding that other issue is located in [this thread ](https://forum.pjrc.com/threads/57609-Teensyduino-1-48-Beta-1?p=216598&viewfull=1#post216598)  
Here is the unedited output from a new ssh session into the same machine without anything stripped out.
`slo@sloNano:~$ arduino --version
Picked up JAVA_TOOL_OPTIONS: 
Set log4j store directory /home/slo/.arduino15
Loading configuration...
Initializing packages...
Preparing boards...
[JmDNS(fe80-0-0-0-5de6-8665-7fcd-96fe-wlan0.local.).Timer] WARN javax.jmdns.impl.tasks.resolver.DNSResolverTask - ServiceResolver(fe80-0-0-0-5de6-8665-7fcd-96fe-wlan0.local.).run() exception 
java.io.IOException: Cannot assign requested address (sendto failed)
	at java.net.PlainDatagramSocketImpl.send(Native Method)
	at java.net.DatagramSocket.send(DatagramSocket.java:693)
	at javax.jmdns.impl.JmDNSImpl.send(JmDNSImpl.java:1618)
	at javax.jmdns.impl.tasks.resolver.DNSResolverTask.run(DNSResolverTask.java:73)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
[fe80-0-0-0-5de6-8665-7fcd-96fe-wlan0.local..recover()] WARN javax.jmdns.impl.JmDNSImpl - RECOVERING
[JmDNS(fe80-0-0-0-5de6-8665-7fcd-96fe-wlan0.local.).State.Timer] WARN javax.jmdns.impl.tasks.state.DNSStateTask - Canceler(fe80-0-0-0-5de6-8665-7fcd-96fe-wlan0.local.).run() exception 
java.io.IOException: Cannot assign requested address (sendto failed)
	at java.net.PlainDatagramSocketImpl.send(Native Method)
	at java.net.DatagramSocket.send(DatagramSocket.java:693)
	at javax.jmdns.impl.JmDNSImpl.send(JmDNSImpl.java:1618)
	at javax.jmdns.impl.tasks.state.DNSStateTask.run(DNSStateTask.java:131)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
Arduino: 1.8.10
java.io.IOException: Stream closed
	at java.io.BufferedInputStream.getBufIfOpen(BufferedInputStream.java:170)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:336)
	at com.fasterxml.jackson.core.json.ByteSourceJsonBootstrapper.ensureLoaded(ByteSourceJsonBootstrapper.java:522)
	at com.fasterxml.jackson.core.json.ByteSourceJsonBootstrapper.detectEncoding(ByteSourceJsonBootstrapper.java:150)
	at com.fasterxml.jackson.core.json.ByteSourceJsonBootstrapper.constructParser(ByteSourceJsonBootstrapper.java:246)
	at com.fasterxml.jackson.core.JsonFactory._createParser(JsonFactory.java:1315)
	at com.fasterxml.jackson.core.JsonFactory.createParser(JsonFactory.java:820)
	at cc.arduino.packages.discoverers.PluggableDiscovery.run(PluggableDiscovery.java:80)
	at java.lang.Thread.run(Thread.java:748)
slo@sloNano:~$ arduino --get-pref sketchbook.path
Picked up JAVA_TOOL_OPTIONS: 
Set log4j store directory /home/slo/.arduino15
Loading configuration...
Initializing packages...
Preparing boards...
[JmDNS(fe80-0-0-0-5de6-8665-7fcd-96fe-wlan0.local.).Timer] WARN javax.jmdns.impl.tasks.resolver.DNSResolverTask - ServiceResolver(fe80-0-0-0-5de6-8665-7fcd-96fe-wlan0.local.).run() exception 
java.io.IOException: Cannot assign requested address (sendto failed)
	at java.net.PlainDatagramSocketImpl.send(Native Method)
	at java.net.DatagramSocket.send(DatagramSocket.java:693)
	at javax.jmdns.impl.JmDNSImpl.send(JmDNSImpl.java:1618)
	at javax.jmdns.impl.tasks.resolver.DNSResolverTask.run(DNSResolverTask.java:73)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
[fe80-0-0-0-5de6-8665-7fcd-96fe-wlan0.local..recover()] WARN javax.jmdns.impl.JmDNSImpl - RECOVERING
[JmDNS(fe80-0-0-0-5de6-8665-7fcd-96fe-wlan0.local.).State.Timer] WARN javax.jmdns.impl.tasks.state.DNSStateTask - Canceler(fe80-0-0-0-5de6-8665-7fcd-96fe-wlan0.local.).run() exception 
java.io.IOException: Cannot assign requested address (sendto failed)
	at java.net.PlainDatagramSocketImpl.send(Native Method)
	at java.net.DatagramSocket.send(DatagramSocket.java:693)
	at javax.jmdns.impl.JmDNSImpl.send(JmDNSImpl.java:1618)
	at javax.jmdns.impl.tasks.state.DNSStateTask.run(DNSStateTask.java:131)
	at java.util.TimerThread.mainLoop(Timer.java:555)
	at java.util.TimerThread.run(Timer.java:505)
/home/slo/Arduino
java.io.IOException: Stream closed
	at java.io.BufferedInputStream.getBufIfOpen(BufferedInputStream.java:170)
	at java.io.BufferedInputStream.read(BufferedInputStream.java:336)
	at com.fasterxml.jackson.core.json.ByteSourceJsonBootstrapper.ensureLoaded(ByteSourceJsonBootstrapper.java:522)
	at com.fasterxml.jackson.core.json.ByteSourceJsonBootstrapper.detectEncoding(ByteSourceJsonBootstrapper.java:150)
	at com.fasterxml.jackson.core.json.ByteSourceJsonBootstrapper.constructParser(ByteSourceJsonBootstrapper.java:246)
	at com.fasterxml.jackson.core.JsonFactory._createParser(JsonFactory.java:1315)
	at com.fasterxml.jackson.core.JsonFactory.createParser(JsonFactory.java:820)
	at cc.arduino.packages.discoverers.PluggableDiscovery.run(PluggableDiscovery.java:80)
	at java.lang.Thread.run(Thread.java:748)
[... 126 lines cut ...]
Picked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
/home/slo/Arduino
[... cut ...]
Global variables use 13396 bytes (4%) of dynamic memory, leaving 314284 bytes for local variables. Maximum is 327680 bytes.
slo@black:~/arduino-1.8.9/arduino-1.8.9$ arduino --verify Blink.inoPicked up JAVA_TOOL_OPTIONS: 
Loading configuration...
Initializing packages...
Preparing boards...
Verifying...
java.io.IOException: No valid code files found
	at processing.app.Sketch.listSketchFiles(Sketch.java:117)
	at processing.app.Sketch.<init>(Sketch.java:54)
	at processing.app.Base.<init>(Base.java:390)
	at processing.app.Base.main(Base.java:144)
`

facchinm, 2019-09-30:
@slomobileAdmin ah now I re-read it and got what your request was. The relative path method requires you to be in the same folder of the .ino (and I believe it's always been like this, for the principle of least astonishment). 
All unix tools work with both absolute and relative paths, but you must pick the relative based on which folder you are into. 
Basing the non-absolute path on the sketchbook would make things very very strange for a CLI program.
I'd close the issue as wontfix; @per1234 @cmaglie do you agree?

slomobileAdmin, 2019-09-30:
@facchinm I do see your point.  Yet the sketchbook as a place to store sketches, as well as the requirement to have a sketch in a folder of the same name is an Arduino convention, not Unix.  Providing the requested functionality need not break relative paths.  Search for absolute and relative paths first.  If they fail to find a candidate, only then search the sketchbook.  It would be like sourcing the sketchbook folder.  I think this will provide a more consistent experience with the gui arduino where File>Open Recent, File>Sketchbook, and File>Examples just list the sketch name and not the path.  It will save newbies some typing and frustration from cryptic exceptions where user intent is very clear.  As the man page points out, the parameter is a filename, not a path.  It should not fail by default if a bare filename is provided.  
If you do indeed close this as wontfix, at least clarify the documentation.

slomobileAdmin, 2019-09-30:
https://linux.die.net/man/3/filename provides the Linux method and precedent to do this.  
value of sketchbook.path is the absolute directory, naked filename parameter is the relative filename.
```
absname(Filename, Dir) -> file:filename()

    Types:

        Filename = file:name()
        Dir = file:filename()

    This function works like absname/1, except that the directory to which the file name should be made relative is given explicitly in the Dir argument.

absname_join(Dir, Filename) -> file:filename()

    Types:

        Dir = file:filename()
        Filename = file:name()

    Joins an absolute directory with a relative filename. Similar to join/2, but on platforms with tight restrictions on raw filename length and no support for symbolic links (read: VxWorks), leading parent directory components in Filename are matched against trailing directory components in Dir so they can be removed from the result - minimizing its length.
```

chrissound, 2019-11-29:
@facchinm surely this error should be reported in a more informative matter? As you mentioned wanting to close this. I received a similar java issue to this.. Not sure if it's related.

closed 2021-04-03 by per1234