arduino/Arduino #9267: Arduino IDE 1.8.10 installation on Linux Mint 19.2

Obamjur, 2019-09-29:
The users must be warned to install <arduino-linux-setup.sh> file after normal installation procedure with <install.sh> file. Update this page https://www.arduino.cc/en/Guide/Linux.

Thanks!

facchinm, 2019-09-30:
@Obamjur running that file is optional since it performs quite some operations on the host PC, so it should be used with care (after reading what it does).
Adding some documentation though wouldn't hurt :wink:

Obamjur, 2019-09-30:
Well, in my case it was mandatory to put Arduino IDE running as expected. I
did not follow documentation comment, so whatever!

per1234, 2021-04-03:
There is some relevant discussion about the script on the PR that added it: https://github.com/arduino/Arduino/pull/6053