arduino/Arduino #9268: GLCD Library for Arduino

DrGamerGuy, 2019-09-29:
Library for using the Graphic LCD type display. Allows displaying of various shapes and images on a Nokia 5110 GLCD screen, may work with other GLCD type screens. Visit javl.github.io/image2cpp/ to make your own image arrays for your screen to display and include them into your program. See Adafruit website for more of their libraries in action. Donwload at https://github.com/DrGamerGuy/GLCD

per1234, 2019-09-30:
Are you requesting that your library be added to the [Arduino Library Manager index](https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ), or are you just advertising your library in our issue tracker?

If you want the library to be added to the index, you'll need to make a change so that it's compliant with our requirements. We require that the library be located in the root of the repository. Your library is in the GLCD subfolder of the repository, which is not allowed. After moving the library, you'll need to either create a new tag/release or redo the existing release. Once that's done, comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

DrGamerGuy, 2019-09-30:
Changes made, you may proceed. Good thing too, one of the files was off and that was also fixed.

cmaglie, 2019-10-03:
done

closed 2019-10-03 by cmaglie