arduino/Arduino #9271: Please, add those libraries to library manager

luisllamasbinaburo, 2019-10-01:
Hi,

Could you please those libraries to the library manager ?
(sorry, I know they are a lot!)

https://github.com/luisllamasbinaburo/Arduino-ReactiveArduino
https://github.com/luisllamasbinaburo/ArduinoLinq
https://github.com/luisllamasbinaburo/Arduino-Interpolation
https://github.com/luisllamasbinaburo/Arduino-Easing
https://github.com/luisllamasbinaburo/Arduino-SyncWaveforms
https://github.com/luisllamasbinaburo/Arduino-Parser
https://github.com/luisllamasbinaburo/Arduino-EasyComma
https://github.com/luisllamasbinaburo/Arduino-BTS7960
https://github.com/luisllamasbinaburo/Arduino-TriangleSolver
https://github.com/luisllamasbinaburo/Arduino-Storyboard
https://github.com/luisllamasbinaburo/Arduino-MultiTask
https://github.com/luisllamasbinaburo/Arduino-AsyncTask
https://github.com/luisllamasbinaburo/Arduino-AsyncTimer
https://github.com/luisllamasbinaburo/Arduino-AsyncSerial
https://github.com/luisllamasbinaburo/Arduino-AsyncServo
https://github.com/luisllamasbinaburo/Arduino-AsyncSonar
https://github.com/luisllamasbinaburo/Arduino-AsyncStepper
https://github.com/luisllamasbinaburo/Arduino-SimpleStepper

Thank you!!

per1234, 2019-10-01:
Hi. Arduino recently started a new policy that 3rd party libraries with a library.properties `name` value starting with the word "Arduino" (or any capitalization thereof) will no longer be accepted for inclusion in the Arduino Library manager index. The reason is that Arduino wants to reserve these names for their own official libraries.

If you want your ArduinoLinq and ArduinoParser libraries to be added to the Library Manager index, you will need to change the `name` value in library properties to one that is compliant with the Arduino Library Specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Then update the `version` value in library.properties and create a new release/tag, or else redo the existing release/tag.

Note that this restriction only applies to the `name` value. You are not required to rename the repository or header files.

Once you have done this, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

luisllamasbinaburo, 2019-10-02:
Ok, sorry. I was not aware of this.

I have already updated both code and releases.

cmaglie, 2019-10-03:
done

closed 2019-10-03 by cmaglie