arduino/Arduino #9273: Arduino 1.8.10 IDE silently fails to open on Windows 10

stockHuman, 2019-10-01:
Hello gang,

Upon installing the latest and greatest IDE, I can no longer start it. Attempting to open it via the executable directly, or though cmd/PS/cygwin does nothing. I have not opened my copy of 1.8.9 in a while, and upon downgrading I now cannot open that either.

I'm currently wondering if this is a Windows 10 thing (I am completely up to date as of time of posting), a dual-monitor thing, or something else entirely.

I have reinstalled the IDE a few times now, if that changes anything. Your help is appreciated.

per1234, 2019-10-01:
Please do this:
1. Start the Windows command line "cmd".
1. Run the program arduino_debug.exe that you will find in the Arduino IDE installation folder from the command line.
1. Copy the output that's printed to the cmd window and paste it here using [code fencing](https://guides.github.com/features/mastering-markdown/).

Hopefully that will give us some information we can use to help you with the problem.

stockHuman, 2019-10-02:
Well there's a handy tool. Thank you for your help!
`-Xmx2048M` was set as a PATH Java option, leading to a failure to allocate the 2097152KB demanded by the IDE. This must be conflicting software or changes I've made an forgotten about since last opening the Arduino.

closed 2019-10-02 by stockHuman

muething, 2020-06-13:
Windows 10 64 bit pro  Updated approx June 8: Tried deleting all and re installing. Here is the results from Command Prompt execution of arduino_debug.exe
Rick Muething, rmuething@cfl.rr.com

c:\Program Files (x86)\Arduino>arduino_debug.exe
Set log4j store directory C:\Users\rmuet\AppData\Local\Arduino15
Loading configuration...
Initializing packages...
java.lang.NullPointerException
        at cc.arduino.contributions.packages.ContributionsIndexer.parseIndex(ContributionsIndexer.java:134)
        at processing.app.BaseNoGui.initPackages(BaseNoGui.java:504)
        at processing.app.Base.<init>(Base.java:273)
        at processing.appBase.manin(Base.java:150)

per1234, 2020-06-13:
Hi @muething. Please see the instructions here: https://github.com/arduino/Arduino/issues/10345