arduino/Arduino #9276: Addition of the BSEC library into the Arduino Library list

BoschSensortec, 2019-10-02:
Hi,

Could you please add the BSEC library to the Arduino Library Manager?

https://github.com/BoschSensortec/BSEC-Arduino-library

Thank you!

dsyleixa, 2019-10-02:
hello,
unfortunately there is no "issues" section available in your github repo.
plese tell how to evaluate for IAQ:

    1 Secondhand smoke
    2 Radon
    3 Molds and other allergens
    4 Carbon monoxide
    5 Volatile organic compounds
    6 Legionella
    7 Other bacteria
    8 Asbestos fibers
    9 Carbon dioxide
    10 Ozone
    1.11 Particulates

per1234, 2019-10-03:
@dsyleixa this is not an appropriate place to ask for support for the BSEC library.

There is a link right in the repository description that points to the support forum. Please use that.

dsyleixa, 2019-10-03:
ok, then I resign because I won't register in an extra forum just for asking for a description in the repo.

cmaglie, 2019-10-03:
The library has been added to the library manager

closed 2019-10-03 by cmaglie