arduino/Arduino #9281: Bug in arduino-linux-setup.sh affecting non-bash users

jwcollins, 2019-10-04:
Hi.  This script has an obvious, but easily fixed bug.  Currently there are a bunch of comments before #!/bin/bash.  And it assumes, very incorrectly, that pound-bang /bin/bash is a comment.  In fact, "#!" is a valid Unix/Linux magic number, and "/bin/bash" is the path to the program which needs to interpret / execute the script.  When run by a user who uses /bin/bash as their everyday shell, it works, but only because the user happens to have invoked bash.  But when run by a user who uses another shell (e.g. /bin/tcsh) as their shell, it doesn't work, because pound-bang is not at the top of the file, and thus /bin/bash is not invoked to execute the script.  And since the script is written to bash, as opposed to tcsh (or ksh or ... ) syntax, it won't work.

This bug report is against distribution 1.8.10.  Screen capture of trying to run this script from a non-bash shell is pasted below.  Read the below wikipedia link, which mis-pronounces "pound-bang", but is otherwise valid.  This construct has worked since handed down by Real Programmer Dennis Ritchie in Jan 1980.  Please learn it.

https://en.wikipedia.org/wiki/Shebang_(Unix)

[image]

endorama, 2019-10-07:
Hi @jwcollins, I provided a PR to fix that.

Meanwhile you can try running the script with explicit interpreter, it should work.  
Ex: `/bin/bash ./arduino-linux-setup.sh`

jwcollins, 2019-10-07:
Thanks Edoardo!  --jwc

PR #9291 (2019-10-10): Moving shebang at the top of the file
Shebang changes how Unix-like OS run scripts, ensuring the script is run with the specified interpreter. In this case as the line was not the first in the file, users without bash as their default shell would experience issues running the script.

Fixes #9281
More details on shebang usage here https://en.wikipedia.org/wiki/Shebang_(Unix)

### All Submissions:

* [x] Have you followed the guidelines in our Contributing document?
* [x] Have you checked to ensure there aren't other open [Pull Requests](https://github.com/arduino/Arduino/pulls?q=) for the same [... cut]
commit 9951c652 (2019-10-10): moving shebang at the top of the file

Shebang changes how Unix-like OS run scripts, ensuring the script is run with the specified interpreter. In this case as the line was not the first in the file, users without bash as their default shell would experience issues running the script.

Fixes #9281
More details on shebang usage here https://en.wikipedia.org/wiki/Shebang_(Unix)
closed 2019-10-10 by facchinm