arduino/Arduino #9287: The arduino IDE can't express file path correctly when it contains a space

Grimoire-of-Chemist, 2019-10-07:
For example, when it comes to "C:/Program Files/...", it displays "C:/Program " instead of a correct one.

cmaglie, 2019-10-07:
Can you provide a screenshot?

Grimoire-of-Chemist, 2019-10-07:
[image]

cmaglie, 2019-10-07:
it may be something related to language settings? can you try, for a moment, to set language to "English" from the preferences and see if the problem persist?

Grimoire-of-Chemist, 2019-10-07:
[image]
It doesn't persist anymore

Grimoire-of-Chemist, 2019-10-07:
But it is quite inconvenient for me to use English 😂

cmaglie, 2019-10-07:
I know, I'm just trying to figure out the cause of the bug :-)

per1234, 2019-10-07:
I've seen this several times on forum output from installations using English language:
- https://forum.arduino.cc/index.php?topic=638123.msg4319562#msg4319562
- https://forum.arduino.cc/index.php?topic=638460
- https://forum.arduino.cc/index.php?topic=639172.msg4326039#msg4326039

ChrisCA, 2019-10-07:
I think I'm running into the same problem and can provide more detailed information.
Windows 10 64-bit, up to date with all patches etc. at the time of writing.
Using a German system environment.
IDE 1.8.10

As is bought a new computer I wanted to setup the Arduino IDE as usual by using the portable ZIP for Windows. When I tried to start the IDE, it closed immediately after showing the splash screen. I started searching around in the web for a while to figgure out what might cause this error which haven't had before. I got a first idea what might cause the error when I started to use the arduino_debug.exe
It reported 

> Sie haben Ihr Sketchbook vergessen: Arduino kann nicht starten, da der Ordner zum
Speichern Ihres Sketchbooks nicht angelegt werden konnte.

Which means translated that my sketchbok is missing and therefore the IDE cannot start. This is somehow strange as the IDE tend to create a new Arduino folder in the Document path automatically if its missing. At this point I already tried several things like compatibility modes, run as admin, the installer version of the IDE, previous version of the IDE and the BETA version, too. Always the same error.
I then simply created the Arduino folder on my Documents folder and the IDE started successfully.
After I could now start the IDE I wanted to install libraries which then failed again with the a long error message which told me that the folder where the library should be installed couldn't be created. I can provide this to you if you need it.

I tried to think about what might be different to my former installations and I assumed that it could be related to my new Windows username respectivley my user folder name which is consting of firstname lastname with a space sign inbetween. Previoulsy I had a username consisting of a single word. I then went to the IDE's preferences and switched my sketch folder to a place which hasn't been affected my username in the path. After doing so, all problems including downloading and installing libraries were solved. For me this seems to be reproducable.

ChrisCA, 2019-10-07:
This is the error message, caused by the IDE after I created the Arduino folder in the Documents directory manually and tried to install a library:

>  C:\Users\Firstname Lastname\Documents\Arduino\libraries
java.lang.RuntimeException: java.nio.file.NoSuchFileException: C:\Users\Firstname Lastname\Documents\Arduino\libraries
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:248)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.nio.file.NoSuchFileException: C:\Users\Firstname Lastname\Documents\Arduino\libraries
	at sun.nio.fs.WindowsException.translateToIOException(WindowsException.java:79)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:97)
	at sun.nio.fs.WindowsException.rethrowAsIOException(WindowsException.java:102)
	at sun.nio.fs.WindowsFileSystemProvider.createDirectory(WindowsFileSystemProvider.java:504)
	at java.nio.file.Files.createDirectory(Files.java:674)
	at java.nio.file.Files.createAndCheckIsDirectory(Files.java:781)
	at java.nio.file.Files.createDirectories(Files.java:767)
	at processing.app.helpers.FileUtils.createTempFolder(FileUtils.java:118)
	at processing.app.helpers.FileUtils.createTempFolder(FileUtils.java:106)
	at processing.app.helpers.FileUtils.createTempFolder(FileUtils.java:102)
	at cc.arduino.contributions.libraries.LibraryInstaller.performInstall(LibraryInstaller.java:169)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:125)
	at cc.arduino.contributions.libraries.LibraryInstaller.install(LibraryInstaller.java:117)
	at cc.arduino.contributions.libraries.ui.LibraryManagerUI.lambda$onInstallPressed$4(LibraryManagerUI.java:239)
	... 1 more

per1234, 2019-12-07:
I think this has now been reported and investigated in detail here: https://github.com/arduino/arduino-cli/issues/493

closed 2020-02-11 by facchinm