arduino/Arduino #9290: Latest Arduino (1.8.10) does not work under the Catalina GM (10.15 Beta 19A582a)

MarqueIV, 2019-10-07:
Just downloaded and upgraded to the GM Release (latest beta, build 19A582a) of 10.15 Catalina along with the latest official release of Arduino (1.8.10).  I'm using an official (i.e. not a clone) Arduino Micro.  Ever since doing so, I can not upload.  The app compiles, but as soon as the upload begins (the light 'breathes' on the device) I get the below.

Note: I also tried the cloud editor with the plugin (again, latest as of tonight) but that didn't work either.  Those results are under these ones.

RESULTS FROM THE ARDUINO IDE:
```
Arduino: 1.8.10 (Mac OS X), Board: "Arduino/Genuino Micro"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/marqueiv/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/marqueiv/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/marqueiv/Code/Arduino/libraries -fqbn=arduino:avr:micro -vid-pid=0X2341_0X8037 -ide-version=10810 -build-path /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795 -warnings=all -build-cache /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_cache_222239 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/marqueiv/Desktop/DeleteMe/Blink/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -hardware /Users/marqueiv/Library/Arduino15/packages -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/marqueiv/Library/Arduino15/packages -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/marqueiv/Code/Arduino/libraries -fqbn=arduino:avr:micro -vid-pid=0X2341_0X8037 -ide-version=10810 -build-path /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795 -warnings=all -build-cache /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_cache_222239 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /Users/marqueiv/Desktop/DeleteMe/Blink/Blink.ino
Using board 'micro' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Detecting libraries used...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8037 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Micro\"" -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/micro /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/sketch/Blink.ino.cpp -o /dev/null
Generating function prototypes...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8037 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Micro\"" -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/micro /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/sketch/Blink.ino.cpp -o /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/preproc/ctags_target_for_gcc_minus_e.cpp
/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -Wall -Wextra -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8037 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Micro\"" -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/micro /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/sketch/Blink.ino.cpp -o /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/sketch/Blink.ino.cpp.o
Compiling libraries...
Compiling core...
Using precompiled core: /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_cache_222239/core/core_arduino_avr_micro_51f02b7210b938436b779d1c032618e1.a
Linking everything together...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-gcc -Wall -Wextra -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega32u4 -o /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/Blink.ino.elf /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/sketch/Blink.ino.cpp.o /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/../arduino_cache_222239/core/core_arduino_avr_micro_51f02b7210b938436b779d1c032618e1.a -L/var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795 -lm
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy -O ihex -j .eeprom --set-section-flags=.eeprom=alloc,load --no-change-warnings --change-section-lma .eeprom=0 /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/Blink.ino.elf /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/Blink.ino.eep
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-objcopy -O ihex -R .eeprom /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/Blink.ino.elf /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/Blink.ino.hex
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-size -A /var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino_build_144795/Blink.ino.elf
Sketch uses 3960 bytes (13%) of program storage space. Maximum is 28672 bytes.
Global variables use 149 bytes (5%) of dynamic memory, leaving 2411 bytes for local variables. Maximum is 2560 bytes.
Forcing reset using 1200bps open/close on port /dev/cu.usbmodem141401
PORTS {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.MALS, /dev/cu.SOC, /dev/cu.usbmodem141401, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.MALS, /dev/tty.SOC, /dev/tty.usbmodem141401, } / {/dev/cu.Bluetooth-Incoming-Port, /dev/cu.MALS, /dev/cu.SOC, /dev/cu.usbmodem141401, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.MALS, /dev/tty.SOC, /dev/tty.usbmodem141401, } => {}
[... cut ...]
[/dev/cu.Bluetooth-Incoming-Port /dev/cu.MALS /dev/cu.SOC /dev/tty.Bluetooth-Incoming-Port /dev/tty.MALS /dev/tty.SOC] -> [/dev/cu.Bluetooth-Incoming-Port /dev/cu.MALS /dev/cu.SOC /dev/cu.usbmodem141401 /dev/tty.Bluetooth-Incoming-Port /dev/tty.MALS /dev/tty.SOC /dev/tty.usbmodem141401]

Found upload port: /dev/tty.usbmodem141401

Flashing with command:/Users/marqueiv/.arduino-create/arduino/avrdude/6.3.0-arduino14/bin/avrdude -C/Users/marqueiv/.arduino-create/arduino/avrdude/6.3.0-arduino14/etc/avrdude.conf -v -patmega32u4 -cavr109 -P/dev/tty.usbmodem141401 -b57600 -D -Uflash:w:/var/folders/xf/2rw7xdq95hl5lctcbjv39w9r0000gn/T/arduino-create-agent151286532/Simple_Playground.hex:i
```

facchinm, 2019-10-07:
It looks like Create is using an old version of avrdude `/Users/marqueiv/.arduino-create/arduino/avrdude/6.3.0-arduino14/bin/avrdude`. 
The Catalina compatible ones are `6.3.0-arduino16` and `6.3.0-arduino17` which you get with AVR core 1.8.1 (bundled in IDE 1.8.10) . May you also paste the verbose output of  the IDE upload phase?

MarqueIV, 2019-10-07:
I believe it's above (the first code block).  The second one was the cloud editor's output.  Lemme make the separation more clear...

facchinm, 2019-10-07:
Ok, great. I may have an idea: it looks like the port reappears with the same name after the 1200bps touch (`/dev/tty.usbmodem141401`) . 
For some reasons maybe the bootloader is not getting recognized as a correct CDC device and the upload starts when the timeout is reached and the sketch restarted.

You may try to keep the board in bootloader mode by pressing the reset button "often" (once every 5 seconds), make sure the LED is fading, check in the meantime if the port is recognized by the OS.

Probably @gvarisco knows some `dmesg`-like command that can help understanding what is going on.

Thanks!

MarqueIV, 2019-10-07:
No, the button just kicks you right out with an error.  Damn.  Well I do have a Windows VM which I'm gonna try to see if that works.  Fingers crossed!  Otherwise I'm DOA!

CSab80, 2019-10-08:
Hello,
Same problem here with Arduino 18.1.11 and Catalina final.
Arduino Nano and Leonardo are concerned.
But, with Windows 7 VM via Parallels Desktop, there is no problem...
By the way, the D13 LED smoothly blinks before the error.
Regards,
Cyril

MarqueIV, 2019-10-09:
Just confirmed my version of macOS is the released version, so this is going to be a critical issue for Mac users as now there's no viable Arduino solution for them short of installing Windows.  Not optimal at all!

gvarisco, 2019-10-09:
We're on it, @MarqueIV. We'll get back to you all once we are done with our findings.

MarqueIV, 2019-10-09:
Sorry.  Just whining because I'm now cut off from one of my most favorite toys! :)

scofano, 2019-10-09:
**Catalina 10.15 (19A583) Final here - I got this from the executable:**

Arduino: 1.8.10 (Mac OS X), Board: "Arduino Leonardo"

Sketch uses 8438 bytes (29%) of program storage space. Maximum is 28672 bytes.
Global variables use 290 bytes (11%) of dynamic memory, leaving 2270 bytes for local variables. Maximum is 2560 bytes.

avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_drain(): read error: Device not configured
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_recv(): read error: Device not configured

Found programmer: Id = ""; type =  
    Software Version = . ; Hardware Version =  . 
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: exit bootloader
avrdude: ser_close(): can't reset attributes for device: Device not configured

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

**And this from the web version:**

/home/builder/opt/arduino-builder/arduino-builder -compile -core-api-version 10611 -hardware /home/builder/opt/arduino-builder/hardware -hardware /home/builder/.arduino15/packages -tools /home/builder/opt/arduino-builder/tools -tools /home/builder/.arduino15/packages -built-in-libraries /home/builder/opt/libraries/latest -logger humantags -fqbn arduino:avr:leonardo -build-cache /tmp -build-path /tmp/921275379/build -verbose -prefs runtime.tools.nrf5x-cl-tools.path=/home/builder/.arduino15/packages/arduino/tools/nrf5x-cl-tools/9.3.1 -prefs runtime.tools.CMSIS-Atmel.path=/home/builder/.arduino15/packages/arduino/tools/CMSIS-Atmel/1.2.0 -prefs runtime.tools.i586-poky-linux-uclibc.path=/home/builder/.arduino15/packages/Intel/tools/i586-poky-linux-uclibc/1.6.2+1.0 -prefs runtime.tools.openocd.path=/home/builder/.arduino15/packages/arduino/tools/openocd/0.10.0-arduino9 -prefs runtime.tools.sketchUploader.path=/home/builder/.arduino15/packages/Intel/tools/sketchUploader/1.6.2+1.0 -prefs runtime.tools.arm-linux-gcc.path=/home/builder/.arduino15/packages/arduino/tools/arm-linux-gcc/4.9.3 -prefs runtime.tools.avr-gcc.path=/home/builder/.arduino15/packages/arduino/tools/avr-gcc/5.4.0-atmel3.6.1-arduino2 -prefs runtime.tools.x86-linux-gcc.path=/home/builder/.arduino15/packages/arduino/tools/x86-linux-gcc/7.2.0 -prefs runtime.tools.esptool.path=/home/builder/.arduino15/packages/esp8266/tools/esptool/2.5.0-3-20ed2b9 -prefs runtime.tools.mkspiffs.path=/home/builder/.arduino15/packages/esp8266/tools/mkspiffs/2.5.0-3-20ed2b9 -prefs runtime.tools.avrdude.path=/home/builder/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino14 -prefs runtime.tools.flashpack.path=/home/builder/.arduino15/packages/Intel/tools/flashpack/2.0.0 -prefs runtime.tools.arduino101load.path=/home/builder/.arduino15/packages/Intel/tools/arduino101load/2.0.1 -prefs runtime.tools.arduinoOTA.path=/home/builder/.arduino15/packages/arduino/tools/arduinoOTA/1.2.1 -prefs runtime.tools.dfu-util.path=/home/builder/.arduino15/packages/arduino/tools/dfu-util/0.9.0-arduino1 -prefs runtime.tools.xtensa-lx106-elf-gcc.path=/home/builder/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/2.5.0-3-20ed2b9 -prefs runtime.tools.bossac.path=/home/builder/.arduino15/packages/arduino/tools/bossac/1.9.1-arduino1 -prefs runtime.tools.linuxuploader.path=/home/builder/.arduino15/packages/arduino/tools/linuxuploader/1.5.1 -prefs runtime.tools.arm-none-eabi-gcc.path=/home/builder/.arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4 -prefs runtime.tools.CMSIS.path=/home/builder/.arduino15/packages/arduino/tools/CMSIS/4.5.0 -prefs runtime.tools.arc-elf32.path=/home/builder/.arduino15/packages/Intel/tools/arc-elf32/1.6.9+1.0.1 -prefs runtime.tools.core2-32-poky-linux.path=/home/builder/.arduino15/packages/Intel/tools/core2-32-poky-linux/1.6.2+1.0 -libraries /tmp/921275379/custom -libraries /tmp/921275379/pinned /tmp/921275379/VA_Robot

[... cut ...]
[/dev/cu.Bluetooth-Incoming-Port /dev/tty.Bluetooth-Incoming-Port] -> [/dev/cu.Bluetooth-Incoming-Port /dev/cu.usbmodem14201 /dev/tty.Bluetooth-Incoming-Port /dev/tty.usbmodem14201]

Found upload port: /dev/cu.usbmodem14201

Flashing with command:/Users/scofano/.arduino-create/arduino/avrdude/6.3.0-arduino14/bin/avrdude -C/Users/scofano/.arduino-create/arduino/avrdude/6.3.0-arduino14/etc/avrdude.conf -v -patmega32u4 -cavr109 -P/dev/cu.usbmodem14201 -b57600 -D -Uflash:w:/var/folders/lb/r402_dh95lv3d43x7k9_f4nm0000gn/T/arduino-create-agent532976100/VA_Robot.hex:i

scofano, 2019-10-09:
**I Tried 1.9.0-beta (desktop) - I'm not spamming, I just want to help with error dumps (in case it helps):**

Arduino: 1.9.0-beta (Mac OS X), Board: "Arduino Leonardo"

Sketch uses 8438 bytes (29%) of program storage space. Maximum is 28672 bytes.
Global variables use 290 bytes (11%) of dynamic memory, leaving 2270 bytes for local variables. Maximum is 2560 bytes.

avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_drain(): read error: Device not configured
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
Found programmer: Id = ""; type =  
    Software Version = . ; Hardware Version =  . 
avrdude: ser_send(): write error: Device not configured
avrdude: ser_send(): write error: Device not configured

avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: exit bootloader
avrdude: ser_close(): can't reset attributes for device: Device not configured

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

scofano, 2019-10-09:
I Tried 1.9.0-beta (desktop) - I'm not spamming, I just want to help with error dumps (in case it helps):

Arduino: 1.9.0-beta (Mac OS X), Board: "Arduino Leonardo"

/Applications/Arduino 2.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino 2.app/Contents/Java/hardware -tools /Applications/Arduino 2.app/Contents/Java/tools-builder -tools /Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino 2.app/Contents/Java/libraries -libraries /Users/scofano/Documents/Arduino/libraries -fqbn=arduino:avr:leonardo -ide-version=10900 -build-path /var/folders/lb/r402_dh95lv3d43x7k9_f4nm0000gn/T/arduino_build_669760 -warnings=none -build-cache /var/folders/lb/r402_dh95lv3d43x7k9_f4nm0000gn/T/arduino_cache_489551 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -verbose -experimental /Users/scofano/Documents/Arduino/VA_01/VA_01.ino
/Applications/Arduino 2.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino 2.app/Contents/Java/hardware -tools /Applications/Arduino 2.app/Contents/Java/tools-builder -tools /Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino 2.app/Contents/Java/libraries -libraries /Users/scofano/Documents/Arduino/libraries -fqbn=arduino:avr:leonardo -ide-version=10900 -build-path /var/folders/lb/r402_dh95lv3d43x7k9_f4nm0000gn/T/arduino_build_669760 -warnings=none -build-cache /var/folders/lb/r402_dh95lv3d43x7k9_f4nm0000gn/T/arduino_cache_489551 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr -verbose -experimental /Users/scofano/Documents/Arduino/VA_01/VA_01.ino
Using board 'leonardo' from platform in folder: /Applications/Arduino
Using core 'arduino' from platform in folder: /Applications/Arduino
Detecting libraries used...
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10900 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\"" "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/variants/leonardo" /var/folders/lb/r402_dh95lv3d43x7k9_f4nm0000gn/T/arduino_build_669760/sketch/VA_01.ino.cpp -o /dev/null
Alternatives for Keyboard.h: [Keyboard@1.0.2]
ResolveLibrary(Keyboard.h)
  -> candidates: [Keyboard@1.0.2]
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10900 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\"" "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/variants/leonardo" "-I/Applications/Arduino 2.app/Contents/Java/libraries/Keyboard/src" /var/folders/lb/r402_dh95lv3d43x7k9_f4nm0000gn/T/arduino_build_669760/sketch/VA_01.ino.cpp -o /dev/null
Alternatives for HID.h: [HID@1.0]
ResolveLibrary(HID.h)
  -> candidates: [HID@1.0]
"/Applications/Arduino 2.app/Contents/Java/hardware/tools/avr/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10900 -DARDUINO_AVR_LEONARDO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8036 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Leonardo\"" "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/cores/arduino" "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/variants/leonardo" "-I/Applications/Arduino 2.app/Contents/Java/libraries/Keyboard/src" "-I/Applications/Arduino 2.app/Contents/Java/hardware/arduino/avr/libraries/HID/src" /var/folders/lb/r402_dh95lv3d43x7 [... cut ...]
avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: leave prog mode
avrdude: ser_send(): write error: Device not configured
avrdude: ser_recv(): read error: Device not configured
avrdude: ser_recv(): read error: Device not configured

avrdude: butterfly_recv(): programmer is not responding
avrdude: error: programmer did not respond to command: exit bootloader
avrdude: ser_close(): can't reset attributes for device: Device not configured

avrdude done.  Thank you.

scofano, 2019-10-10:
Unfortunately, this new build didn't work (pasted the log on #9303 )
Please, let me know if you need to test anything else.

nooges, 2019-10-11:
I've run into the same issue like @facchinm observed with the 1200 baud touch not doing anything and noted it here: https://github.com/qmk/qmk_firmware/issues/6133#issuecomment-531884906

I'm seeing `mctl RS232_S_OUTPUTS failed` when I check things in `dmesg`.

d53dave, 2019-10-11:
I'm seeing the exact same behavior, unfortunately.

dlwalter, 2019-10-11:
I am seeing this as well with the 1.9.x beta on Catalina.

jmiguelff, 2019-10-12:
Same issue... Looks like an issue with AVRDude trying to communicate with the serial interface? Trying to update a project to Arduino Leonardo using macOS Catalina 10.15.
Just another hint, the "Get Board Info" functionality works...

MicroHydroMan, 2019-10-12:
Have not been around Arduino in some time, just trying to get back into a sketch - turbine control application I wrote a few years ago. "User configuration file does not exist... " when I upload the simplest sketch.  Mac OS 10.15, Arduino 1.8.10.
I broke out the old Powerbook G4 OS 10.4.11 Arduino 1.0.6 where the original sketch was developed, compiled and uploaded successfully years ago. Now any upload attempts end with 
 " avrdude: stk500_getsync(): not in sync: resp=0x30"
What is going on?  It used to be easy. Please help me get started again.

Pimvandermeer, 2019-10-12:
For those who don't want to downgrade their OS but still want to continue with Arduino IDE workaround is installing a windows VM.

gvarisco, 2019-10-13:
We're very sorry this is happening, looks like macOS Catalina update broke the bootloader recognition for atmega32u4-based Arduino boards. We've filled a case (FB7376052 for reference) with Apple and we hope to hear back from them as this is impacting lot of users.

CSab80, 2019-10-13:
> We're very sorry this is happening, looks like macOS Catalina update broke the bootloader recognition for atmega32u4-based Arduino boards. We've filled a case (FB7376052 for reference) with Apple and we hope to hear back from them as this is impacting lot of users.

Hello,
Thank you very much for your feedback. We'll gonna be patient.
By the way, and only for my curiosity, why upload works via a VM ?
Regards,
Cyril

MarqueIV, 2019-10-15:
> Hello,
> Thank you very much for your feedback. We'll gonna be patient.
> By the way, and only for my curiosity, why upload works via a VM ?
> Regards,
> Cyril

They work via a VM because the VM is a different operating system.  The issue is with Catalina, not a machine.

CSab80, 2019-10-15:
> > By the way, and only for my curiosity, why upload works via a VM ?
> 
> Are you saying they work when Catalina is run under a VM? That would be interesting.
> 
> However if you're asking why, as others have stated, you can do this under a VM running Windows or even earlier versions of macOS, it's because exactly of that... those are different OS versions. The issue is with Catalina, not the computer or the Arduino.

Yes, that's what I say... :-)

With Windows 7 under Parallels Desktop, Arduino IDE uplaods to Leonardo work correctly.

By the way, the last Catalina update (macOS 10.15 (19A602)) doesn't change anything.

Regards,

Cyril

w-A-L-L-e, 2019-10-16:
The issue is indeed with catalina. It works on my same laptop with triple boot : the windows and linux installed on it still flash .hex files perfectly with avrdude. Only macos catalina is broken. Other things broken: mini display ports no longer support higher resolutions anymore (for instance external monitor used to be able to do 3840x2160 both with hdmi and display port on mojave. after catelina update the higher resolution only works on the big hdmi port. But no longer on the mini ports they now fall back to 1920x1024) So basically a lot of stuff is now broken on Catalina, wished I stayed on Mojave now...

metroyanno, 2019-10-16:
Have the same issue. And that happened after the update. Tried to connect other serial devices or redirect to Parallels Desktop on Windows, unsuccessfully. I think that it is a reason in OS X

facchinm, 2019-10-16:
Hi everyone,
after some tests on real hardware it appears that the issue is about the Caterina bootloader being recognized as "something else" (see https://github.com/qmk/qmk_firmware/issues/6133#issuecomment-542147187)

I'm trying to find the specific problem that triggers that behaviour to speedup Apple fix.

In the meantime, the experimental Caterina2 bootloader (from [this](https://github.com/facchinm/Arduino_avrusb_firmware) repo) works on Catalina.

If you have an external debugger or another Arduino board not affected by the bug (like a UNO) you can flash it using the attached binary and these command lines
```
.../6.3.0-arduino17/bin/avrdude -C.../6.3.0-arduino17/etc/avrdude.conf -v -patmega32u4 -catmelice_isp -Pusb -e -Ulock:w:0x3F:m -Uefuse:w:0xfb:m -Uhfuse:w:0xd8:m -Ulfuse:w:0xff:m
.../6.3.0-arduino17/bin/avrdude -C.../6.3.0-arduino17/etc/avrdude.conf -v -patmega32u4 -catmelice_isp -Pusb -Uflash:w:Caterina2-Yun.hex:i -Ulock:w:0x2F:m
```
You need to change `...` with the actual path of your avrdude and `-catmelice_isp` with your programmer.
The first command removes the bootloader flash protection, while the second one flashes the real bootloader. On a successful load, the fading led will be faster than previously.

Good guide on ArduinoAsISP: https://www.microcontrollertips.com/programming-avr-microcontrollers-arduinos-arduino-isp/

ArduinoAsISP example (also included in the IDE): https://github.com/arduino/Arduino/blob/master/build/shared/examples/11.ArduinoISP/ArduinoISP/ArduinoISP.ino

[Caterina2.zip](https://github.com/arduino/Arduino/files/3734610/Caterina2.zip)

jhoughjr, 2019-10-16:
I'd like to add as it is maybe relevant that this also happens using Arduino with esptool.py instead of avrdude. I also just updated my SILabs drivers as well. Funny thing is this worked for me all through the Catalina betas.I'll reboot soon and see if that helps re the driver install though it didnt say it required restart.

zixzixzix, 2019-10-17:
So what to do to fix thIs bug?
The only way is waiting apple to fix the Problem?

jmiguelff, 2019-10-17:
You can do what @facchinm said, which is upgrade the bootloader of the microprocessor or wait for Apple to change their way of handling baudrate changes.
I will try to do upgrade the bootloader as I don't think Apple will be convinced that changing the baudrate should only raise a warning and not disconnect the device.

By the way great work @facchinm !

harbaum, 2019-10-17:
Would it be possible to directly access the bootloader using something like libusb? That way a modified avrdude could be able to bypass the macos acm layer.

rolzing, 2019-10-17:
do we need to create a post on apple forums ?

gvarisco, 2019-10-17:
> do we need to create a post on apple forums ?

Apple forums might help, but I do suggest to file a bug using their new 'Feedback Assistant' tool: https://feedbackassistant.apple.com

w-A-L-L-e, 2019-10-17:
@rolzing indeed, ideally apple should fix it as it means currently bricking all existing arduino leonardo or compatible devices (basically 90% of atmega32u4 devices in the wild). 

It requires reprogramming a new bootloader for these in order to get them working again (allowing uploading of new sketches from catalina or firmware upgrades in our case also are broken on catalina due to this). I've applied the patch from @facchinm to our bootloader as well and recompiled a .hex of anykey bootloader with this patch (the anykey bootloader is useable for other arduinos, it just adds a copy+write protection by allowing to turn on/off the bootloader using a flag in eeprom). 

Will test this on mac + win + linux before merging into master. https://github.com/w-A-L-L-e/anykey-bootloader/blob/development/anykey/anykey_bootloader.hex 
Thanks @facchinm for the quick patch!

jhoughjr, 2019-10-17:
if this is due to an exception due to baud rate change failure apple isnt likely to change that.
The exception should be handled by the caller.

zixzixzix, 2019-10-20:
Where is Caterina.c ?How could i edit it? i could not find it on Arduino  for macOS

sashok1337, 2019-10-20:
@zixzixzix it's a part of [ArduinoCore-avr](https://github.com/arduino/ArduinoCore-avr/blob/master/bootloaders/caterina/Caterina.c)

triplepoint, 2019-10-20:
If anyone finds themselves in this ticket's situation:
- Running MacOS Catalina
- Don't want to (or can't) downgrade to an older MacOS verson
- Don't have access to (or don't want to) patch the bootloader on your device

I can confirm that running Arduino in a virtualbox instance can work.  However you end up opening a can of not insignificant worms around USB access between the host (MacOS) and the guest OS.

I got something working for my own use case, and wanted to share a heavily-commented `Vagrantfile` which should get you rolling with `arduino-cli`.  This was sufficient for me to compile and load my Adafruit Feather 32u4, and I have no reason to believe it wouldn't work for most other hardware as well.

https://gist.github.com/triplepoint/3d6d6070dc9b974b2a715d4dc2f2f59e

Good luck.

fubyo, 2019-10-21:
Isn't there a way to directly patch AVRDude, so that it handshakes exactly the same way virtualbox is doing it?

jmiguelff, 2019-10-21:
I am using Caterina2 bootloader as @facchinm suggested and it works as expected. Anyone knows when are the boards going to be released with this new bootloader?

txs72, 2019-10-21:
Thanks @facchinm. The patch works well on my Catalina v10.15 19A602.

p1ne, 2019-10-23:
Here I created Vagrantfile which does everything via provisioning. One can use it for sample. My project is based on Spark Fun Pro Micro, that's why VID/PID are set for it. PlatformIO used as backend

zixzixzix, 2019-10-23:
I think i need to compile  BootLoader file.
I think its too far for me.  And my English is not good.
When a run make.
`/bin/sh: avr-size: command not found
/bin/sh: avr-size: command not found
 [INFO]    : Begin compilation of project "Caterina2"...

make: avr-gcc: No such file or directory
make: *** [build_begin] Error 1`
Could anyone upload the hex file ?

CSab80, 2019-10-23:
> I think i need to compile BootLoader file.
> I think its too far for me. And my English is not good.
> When a run make.
> `/bin/sh: avr-size: command not found
> /bin/sh: avr-size: command not found
> [INFO] [...]

At the bottom of the post there is a link.
https://github.com/arduino/Arduino/issues/9290#issuecomment-542706407

Regards.

zixzixzix, 2019-10-23:
Thank you all!
It works

w-A-L-L-e, 2019-10-23:
@zixzixzix https://github.com/w-A-L-L-e/anykey-bootloader/blob/development/anykey/anykey_bootloader.hex

CSab80, 2019-10-24:
Rrrrrr !!! I'm going to be mad ! All is done from Catalina.

I'm using a Uno as programmer. The target is a Leonardo, of course. I've injected the sketch ArduinoISP in the Uno, then I'have wired the Uno to the Leonardo.

I've tried the two ways : to the ICSP of the Leo, and to pins 11, 12, and 13.

Here is my command line :
`/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avrdude -C /Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf -v -p atmega32u4 -c arduino -P /dev/cu.usbserial-141230 -e -U lock:w:0x3F:m -U efuse:w:0xfb:m -U hfuse:w:0xd8:m -U lfuse:w:0xff:m`

And here the output :
`avrdude: Version 6.3-20190619
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Applications/Arduino.app/Contents/Java/hardware/tools/avr/etc/avrdude.conf"
         User configuration file is "/Users/csab/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-141230
         Using Programmer              : arduino
         AVR Part                      : ATmega32U4
         Chip Erase delay              : 9000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
           flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 4.4
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e950f (probably m328p)
avrdude: Expected signature for ATmega32U4 is 1E 95 87
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.
`

Why avrdude see the Uno's chip, and not the Leo one ???

I'm going to try via a Linux.

Regards,

Cyril

akamensky, 2019-10-30:
I've got an express M4, thinking whether I could use that to write patched bootloader. But not sure what the correct pinout for the default sketch would be. Do I only need to assign one GPIO that connects to reset pin? Can I pick any?

beerreason, 2019-10-30:
For what it's worth, I updated my Macbook to 10.15.1 last night, and I'm still getting the same failure today.

ladyada, 2019-10-30:
hiya if someone is enterprising they can update 
https://github.com/adafruit/Adafruit_AVRProg
to have a sketch for the leonardo bootloader, this library runs on any microcontroller

jhoughjr, 2019-10-30:
Anyone else still unable to use Arduino on esp? esptool.py 2.6 that seems to be included in Arduino might be the problem. I installed 2.7 which may have a fix, but Arduino still reports 2.6. Point being there must be other Catalina issues than the avr boot loader as esps would use something different? I've been down for well over a week now. Everything works fine on windows 10 same version of Arduino.

jhoughjr, 2019-10-30:
> For what it's worth, I updated my Macbook to 10.15.1 last night, and I'm still getting the same failure today.

I'm digging deeper myself as I can't wait for the community to fix it it seems.

I'm going to try copying esptool.py into the location of 2.6.1 in library/Arduino15 
if you aren't using ESP, perhaps doing the same for updating avrdude will fix it for Catalina also.
I haven't tried burning to an avr in a while. If this works I'll let you know with instructions so you can try it.

akamensky, 2019-10-31:
Is there a patched bootloader for Arduino feather 32u4? I've successfully setup to use M4 express feather as ArduinoAsISP programmer, but all of the bootloaders attached here don't seem to be working (32u4 usb is not enumerating and it is flashing red led #13 rapidly).

w-A-L-L-e, 2019-10-31:
@akamensky have you tried ours yet? https://github.com/w-A-L-L-e/anykey-bootloader/blob/development/anykey/anykey_bootloader.hex it's been tested on a 32u4 but not a bluefruit yet.

akamensky, 2019-10-31:
@w-A-L-L-e yes, one of the tested. The only way I get the board back is by doing "Burn bootloader" from Arduino IDE, which restores it to original state.

I am trying now to find all parts needed to compile bootloader from source (applying patch above), but this is also a challenge since there is 0 documentation and pieces are missing and require a lot of guesswork :sigh:

akamensky, 2019-10-31:
After struggling with 0 documentation, I've managed to compile a bootloader for Feather32u4 board with patch from https://github.com/arduino/Arduino/issues/9290#issuecomment-542728021. Amazingly this works like a charm on macos Catalina now. Flashing any code now goes without a problem:

[Patched Caterina-Feather32u4.hex.zip](https://github.com/arduino/Arduino/files/3794764/Caterina-Feather32u4.hex.zip)

Compressed instructions to compiling a bootloader on macos (assuming patch already applied):

**Standard warning: Here be dragons, RTFM, use with caution, your mileage may vary and not my responsibility.**
```
1. Install Arduino IDE and download board files from "Boards Manager"
2. Download untainted LUFA-111009 (i.e. from https://github.com/reeedstudio/Seeed32U4Bootloader) and put it under ~/Library/Arduino15/packages/LUFA
3. Open ~/Library/Arduino15/packages/adafruit/hardware/avr/1.4.13/boards.txt
4. Locate section for your board and take a note of "build.board", "build.f_cpu", "build.vid" and "build.pid"
5. $ brew tap osx-cross/avr && brew install avr-gcc@4
6. $ export PATH="/usr/local/opt/avr-gcc@4/bin:$PATH"
7. $ cd ~/Library/Arduino15/packages/adafruit/hardware/avr/1.4.13/bootloaders/caterina
8. $ make clean && make BOARD=${build.board} F_CPU=${build.f_cpu} VID=${build.vid} PID=${build.pid} all
9. cp Caterina.hex %your_board_hex_file%.hex
10. Can do Burn Bootloader via Arduino IDE
```

akamensky, 2019-10-31:
@w-A-L-L-e maybe I missed something, I've tried just that and it was not booting.

jmiguelff, 2019-10-31:
@akamensky you should notice Adafruit about your work on the bootloader! Looks cool :D

Should this issue be closed with the solution being update the bootloader?

michelecos, 2019-11-01:
I am running 1.8.11 hourly build 2019/10/29 11:12. I have been able to blink the yellow led effortlessly :-).
My board is an Arduino UNO R3 from an Elegoo kit, connected on USB on a Macbook Pro 15. OS version is macOS 10.15 (19A602).

w-A-L-L-e, 2019-11-01:
@akamensky that's odd, the latest anykey_bootloader.hex works fine here on catalina and win10 with our 32u4 based boards. Don't have a feather32u4 here to replicate the issue. Anyway, great news you got it working too for your board too.

d53dave, 2019-11-01:
> My board is an Arduino UNO R3 from an Elegoo kit, connected on USB on a Macbook Pro 15. OS version is macOS 10.15 (19A602).

The issue in this thread is affecting 32u4 based boards.

w-A-L-L-e, 2019-11-01:
@michelecos arduino uno is not using a software usb bootloader but a dedicated serial->usb chip for programming. The issue is only with the software usb bootloaders (those boards have advantage of coming in smaller boards and allow updating + protecting the bootloader as well and can emulate a usb mouse or keyboard) . An example is the arduino leonardo, but there are literally thousands of derivatives and clones that basically all broke with the Catalina update and need to patch their bootloaders with above described patches.

[47 more replies]

closed 2020-01-09 by gvarisco