arduino/Arduino #9299: [Library Manager] Add Adeon to Library Manager

vzahradnik, 2019-10-09:
Hello,
please add our library into your library manager.

Link to the repository:
https://github.com/JSC-electronics/Adeon

And to library.properties file:
https://github.com/JSC-electronics/Adeon/blob/master/library.properties

Thanks!

per1234, 2019-10-11:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

Ales-Svoboda, 2019-10-12:
Hello,
the release has been created.

Kind regards,
AS.

cmaglie, 2019-10-16:
Done.

closed 2019-10-16 by cmaglie