arduino/Arduino #9304: Si4844 Library for Arduino

pu2clr, 2019-10-10:
https://github.com/pu2clr/SI4844

pu2clr, 2019-10-10:
This is a library for the SI4844, BROADCAST ANALOG TUNING DIGITAL DISPLAY AM/FM/SW RADIO RECEIVER, IC from Silicon Labs for the Arduino development environment. This library is intended to provide an easier interface for controlling the SI4844.

By Ricardo Lima Caratti, Oct, 2019.

per1234, 2019-10-11:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

pu2clr, 2019-10-11:
Thanks for the answer. The release was created.