arduino/Arduino #9308: dtostrf is not avaiable for ARM Cortex M4

dsyleixa, 2019-10-11:
dtostrf is not avaiable for ARM Cortex M4.
I'm using IDE 1.8.9, 
Arduino SAMD Boards M0+ 1.8.4
Adafruit SAMD Boards 1.5.4 

`D:\arduino\portable\sketchbook\libraries\MPU6050\MPU6050.cpp:3308:197: error: 'dtostrf' was not declared in this scope`

how to fix that?

```
Arduino: 1.8.9 (Windows 7), Board: "Adafruit Feather M4 Express (SAMD51), Enabled, 120 MHz (standard), Small (-Os) (standard), 50 MHz (standard), Arduino, Off"

D:\arduino\arduino-builder -dump-prefs -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=adafruit:samd:adafruit_feather_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=0X239A_0X8022 -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_364825 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.openocd.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.openocd-0.9.0-arduino.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0.path=D:\arduino\portable\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.CMSIS-Atmel.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -verbose C:\Users\hw\AppData\Local\Temp\arduino_modified_sketch_209991\MPU6050_DMP6_noIRQ.ino
D:\arduino\arduino-builder -compile -logger=machine -hardware D:\arduino\hardware -hardware D:\arduino\portable\packages -tools D:\arduino\tools-builder -tools D:\arduino\hardware\tools\avr -tools D:\arduino\portable\packages -built-in-libraries D:\arduino\libraries -libraries D:\arduino\portable\sketchbook\libraries -fqbn=adafruit:samd:adafruit_feather_m4:cache=on,speed=120,opt=small,maxqspi=50,usbstack=arduino,debug=off -vid-pid=0X239A_0X8022 -ide-version=10809 -build-path C:\Users\hw\AppData\Local\Temp\arduino_build_364825 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.CMSIS.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.CMSIS-4.5.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS\4.5.0 -prefs=runtime.tools.arm-none-eabi-gcc.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.arm-none-eabi-gcc-4.8.3-2014q1.path=D:\arduino\portable\packages\arduino\tools\arm-none-eabi-gcc\4.8.3-2014q1 -prefs=runtime.tools.openocd.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.openocd-0.9.0-arduino.path=D:\arduino\portable\packages\arduino\tools\openocd\0.9.0-arduino -prefs=runtime.tools.bossac.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -prefs=runtime.tools.bossac-1.7.0.path=D:\arduino\portable\packages\arduino\tools\bossac\1.7.0 -prefs=runtime.tools.CMSIS-Atmel.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.CMSIS-Atmel-1.2.0.path=D:\arduino\portable\packages\arduino\tools\CMSIS-Atmel\1.2.0 -prefs=runtime.tools.bossac-1.8.0-48-gb176eee.path=D:\arduino\portable\packages\arduino\tools\bossac\1.8.0-48-gb176eee -verbose C:\Users\hw\AppData\Local\Temp\arduino_modified_sketch_209991\MPU6050_DMP6_noIRQ.ino
Using board 'adafruit_feather_m4' from platform in folder: D:\arduino\portable\packages\adafruit\hardware\samd\1.5.4
Using core 'arduino' from platform in folder: D:\arduino\portable\packages\adafruit\hardware\samd\1.5.4
Build-Optionen wurden verändert, alles wird neu kompiliert
Detecting libraries used...
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"MPU6050_DMP6_noIRQ.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10809 -DARDUINO_FEATHER_M4 -DARDUINO_ARCH_SAMD -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\cores\\arduino/Adafruit_TinyUSB_Core" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\cores\\arduino/Adafruit_TinyUSB_Core/tinyusb/src" -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE -Os -DVARIANT_QSPI_BAUD_DEFAULT=50000000 "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\cores\\arduino" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\variants\\feather_m4" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_364825\\sketch\\MPU6050_DMP6_noIRQ.ino.cpp" -o nul
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"MPU6050_DMP6_noIRQ.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10809 -DARDUINO_FEATHER_M4 -DARDUINO_ARCH_SAMD -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\cores\\arduino/Adafruit_TinyUSB_Core" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\cores\\arduino/Adafruit_TinyUSB_Core/tinyusb/src" -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE -Os -DVARIANT_QSPI_BAUD_DEFAULT=50000000 "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\cores\\arduino" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\variants\\feather_m4" "-ID:\\arduino\\portable\\sketchbook\\libraries\\I2Cdev" "C:\\Users\\hw\\AppData\\Local\\Temp\\arduino_build_364825\\sketch\\MPU6050_DMP6_noIRQ.ino.cpp" -o nul
"D:\\arduino\\portable\\packages\\arduino\\tools\\arm-none-eabi-gcc\\4.8.3-2014q1/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -c -Os -w -std=gnu++11 -ffunction-sections -fdata-sections -fno-threadsafe-statics -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions "-D__SKETCH_NAME__=\"\"\"MPU6050_DMP6_noIRQ.ino\"\"\"" -w -x c++ -E -CC -DF_CPU=120000000L -DARDUINO=10809 -DARDUINO_FEATHER_M4 -DARDUINO_ARCH_SAMD -D__SAMD51J19A__ -DADAFRUIT_FEATHER_M4_EXPRESS -D__SAMD51__ -DUSB_VID=0x239A -DUSB_PID=0x8022 -DUSBCON -DUSB_CONFIG_POWER=100 "-DUSB_MANUFACTURER=\"Adafruit LLC\"" "-DUSB_PRODUCT=\"Adafruit Feather M4\"" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\cores\\arduino/Adafruit_TinyUSB_Core" "-ID:\\arduino\\portable\\packages\\adafruit\\hardware\\samd\\1.5.4\\cores\\arduino/Adafruit_TinyUSB_Core/tinyusb/src" -D__FPU_PRESENT -DARM_MATH_CM4 -mfloat-abi=hard -mfpu=fpv4-sp-d16 -DENABLE_CACHE -Os -DVARIANT_QSPI_BAUD_DEFAULT=50000000 "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS\\4.5.0/CMSIS/Include/" "-ID:\\arduino\\portable\\packages\\arduino\\tools\\CMSIS-Atmel\\1.2.0/CMSIS/Device/ATMEL/ [... cut ...]
Bibliothek I2Cdev im Ordner: D:\arduino\portable\sketchbook\libraries\I2Cdev (legacy) wird verwendet
Bibliothek Wire in Version 1.0 im Ordner: D:\arduino\portable\packages\adafruit\hardware\samd\1.5.4\libraries\Wire  wird verwendet
Bibliothek MPU6050 im Ordner: D:\arduino\portable\sketchbook\libraries\MPU6050 (legacy) wird verwendet
Bibliothek elapsedMillis in Version 1.0.5 im Ordner: D:\arduino\portable\sketchbook\libraries\elapsedMillis  wird verwendet
exit status 1
Fehler beim Kompilieren für das Board Adafruit Feather M4 Express (SAMD51).
```

matthijskooijman, 2019-10-11:
This function is not part of the Arduino API, nor of the standard C API AFAIK, I think it is just a AVR-libc addition that is just somewhat convenient and thus used. There should be alternatives to use (possibly functions that are not available on AVR, such as `sprintf` with float support). I think this issue has come up on this tracker (or maybe the arduinocore-sam repository) before, so maybe you could find some suggestions in an earlier issue: 
https://github.com/arduino/Arduino/issues?q=is%3Aissue+sort%3Aupdated-desc+dtostrf+is%3Aclosed

Also, please search for issues (including closed ones) before reporting a new one in the future. I'm closing this one as duplicate.

dsyleixa, 2019-10-11:
yes, it seems to be in the standard AVR-libc which is installed by Arduino by default and thus available by default.
But as Arduino supports also other cores than AVR, this AVR-libc must be available also to all other cores by default.

dsyleixa, 2019-10-11:
BTW, I have been searching for 
dtostrf ARM Cortex
but didn't find resolved issues

dsyleixa, 2019-10-11:
BTW, I have been searching for
dtostrf ARM Cortex
but didn't find resolved issues (still not yet, because as stated, it should be available by default for all cores just like for AVRs) - so it's NOT a duplicate actually.

per1234, 2019-10-11:
>this AVR-libc must be available also to all other cores by default.

And it is. You just need to add this:
```c++
#include <avr/dstrtof.h>
```
to your program.

>I have been searching for dtostrf ARM Cortex but didn't find resolved issues

I'm amazed by that, because when I click the link provided by matthijskooijman I find the answer in the very first result: https://github.com/arduino/Arduino/issues/3411

In the future, when you have an issue to report that is specific to the Arduino SAMD Boards core, please do it in the correct repository:
https://github.com/arduino/ArduinoCore-samd
(but search this issue tracker also first because there are many other people like you who don't use the correct issue tracker).

This issue tracker is only to be used for reporting issues related to the Arduino IDE or issues that don't match with any other official Arduino repository.

>I have been searching for dtostrf ARM Cortex

You're poisoning your search by adding the completely unnecessary "ARM Cortex" keywords.

closed 2019-10-11 by per1234

dsyleixa, 2019-10-11:
> You're poisoning your search by adding the completely unnecessary "ARM Cortex" keywords.

no, it's just what I am searching for: ARM Cortex!

per1234, 2019-10-11:
Those terms are almost never used in the Arduino world so you are indeed poisoning your search. You need to work on improve your search skills. It's an essential skill to have in this day and if you want to be a positive part of this issue tracker.

Locking this thread since it's no longer a productive discussion.