arduino/Arduino #9309: error when including  <avr/dstrtof.h>

dsyleixa, 2019-10-11:
error when including 
#include <avr/dstrtof.h>
(as suggested in https://github.com/arduino/Arduino/issues/9308#issuecomment-541160600 )

 exit status 1 avr/dstrtof.h: No such file or directory

BTW, it's rather narrow-minded to close the discusion at that point of discussion.

dsyleixa, 2019-10-11:
probably a typo

#include <avr/dtostrf.h>

dsyleixa, 2019-10-11:
but even then the error stays the same:

#include <avr/dtostrf.h>
...

:\arduino\portable\sketchbook\libraries\MPU6050\MPU6050.cpp:3308:197: error: 'dtostrf' was not declared in this scope

cmaglie, 2019-10-11:
Did you add the suggested include in `arduino\portable\sketchbook\libraries\MPU6050\MPU6050.cpp\MPU6050.cpp`?
Probably the library has only been tested on AVR and doesn't include that file.

dsyleixa, 2019-10-11:
no, I added it just in my code, never thought about having to add a library more than just once

dsyleixa, 2019-10-11:
thanks, that solved it!
I appreciate this kind of constructive conversation!

closed 2019-10-11 by dsyleixa