arduino/Arduino #9314: macOS signing certificate for IDE and Arduino Creator Web Plugin not valid on Catalina

docswab, 2019-10-13:
The released 1.8.10 and 19 September hourly build of Arduino IDE and the Arduino Create Agent 1.1 for macOS cannot be opened on the release version of  Catalina 10.15 because "Apple cannot check it for malicious software."

The certificate and notarization needs to be updated.

docswab, 2019-10-13:
Additionally, the 1.9 beta build 115 for MacOS also will not open. It states that "macOS cannot verify that this app is free from malware."

gvarisco, 2019-10-13:
Hi @docswab, thanks for opening this issue! Below the current situation:

- IDE 1.8.10 on macOS Catalina is properly signed and is expected to be working by 'Show in Finder' -> 'right-clicking' on the application icon -> 'Open'. We'll have to expedite notarization to make it working without having the following pop-up to appear (cc @facchinm).

[image]

>  % codesign -dv --verbose=2 /Applications/Arduino.app
Executable=/Applications/Arduino.app/Contents/MacOS/Arduino
Identifier=cc.arduino.Arduino
Format=app bundle with Mach-O universal (i386 x86_64)
CodeDirectory v=20200 size=466 flags=0x0(none) hashes=9+3 location=embedded
Library validation warning=OS X SDK version before 10.9 does not support Library Validation
Signature size=8907
**Authority=Developer ID Application: ARDUINO SA (7KT7ZWMCJT)
Authority=Developer ID Certification Authority
Authority=Apple Root CA**
Timestamp=13 Sep 2019 at 12:24:46
Info.plist entries=26
TeamIdentifier=7KT7ZWMCJT
Sealed Resources version=2 rules=13 files=4721
Internal requirements count=1 size=180  

- Nightly/Beta are not signed and therefore aren't running any longer on macOS Catalina. This is something we shall definitely address. I'll pass this to our dev team (cc @facchinm, @rsora) for further review.

- Arduino Create Agent 1.1 installers (ArduinoCreateAgent-1.1-osx-installer-chrome.dmg, ArduinoCreateAgent-1.1-osx-installer-firefox.dmg, ArduinoCreateAgent-1.1-osx-installer-safari.dmg) should be working as Developer ID passes notarization requirements:

> % spctl -a -t open --context context:primary-signature -v ArduinoCreateAgent-1.1-osx-installer-*
ArduinoCreateAgent-1.1-osx-installer-chrome.dmg: accepted
source=Notarized Developer ID
ArduinoCreateAgent-1.1-osx-installer-firefox.dmg: accepted
source=Notarized Developer ID
ArduinoCreateAgent-1.1-osx-installer-safari.dmg: rejected
source=Unnotarized Developer ID 

Despite not being cited in their [official docs](https://developer.apple.com/documentation/security/notarizing_your_app_before_distribution), the developer's announcement published [here](https://developer.apple.com/news/?id=09032019a) relaxed notarization prerequisites until January 2020.

Also here, right-clicking on the installer -> 'Open' may be needed. We'll expedite notarization as soon as possible. /cc @rsora

facchinm, 2019-10-14:
I'm not a huge fan of signing prerelease / beta software (generated from a CI) since it gives it a bit of "officiality" that it probably doesn't deserve :slightly_smiling_face: 
Said that, if it's really impossible to open the Beta / Houly on Catalina we can think about something.

docswab, 2019-10-20:
Thank you. That does work for me. I appreciate your help. 

Yours,
Steve Allison-Bunnell

> On Oct 13, 2019, at 3:25 PM, Gianluca Varisco <notifications@github.com> wrote:
> 
> IDE 1.8.10 on macOS Catalina is properly signed and is expected to be working by 'Show in Finder' -> 'right-clicking' on the application icon -> 'Open'.

ryandary, 2019-12-06:
I found this issue through one of the closed issues where this was noted as the issue those ones duplicated.  I just wanted to provide some additional feedback about this as I would like at least the official builds to be signed and notarized so that they function well with the macOS Catalina security settings.  Thanks.

baplacador, 2019-12-17:
Great! it helps a lot.
 
 On Oct 13, 2019, at 3:25 PM, Gianluca Varisco ***@***.***> wrote:

 IDE 1.8.10 on macOS Catalina is properly signed and is expected to be working by 'Show in Finder' -> 'right-clicking' on the application icon -> 'Open'.

cmaglie, 2020-03-20:
This issue has been fixed.

closed 2020-03-20 by cmaglie

mlromramse, 2020-08-19:
I was trying to accomplish install via arduino.cc/download but to no avail. I read a lot of forums like this but it didn't help.

My suggestion is to skip their download/install completely and install homebrew and install arduino from there instead. Completely without aches.

Install homebrew from here: https://brew.sh/

Install Arduino.app using homebrew in your terminal by typing this: brew cask install arduino

Happy coding