arduino/Arduino #9322: [Library Manager] Please add desklab Library

jnsdrtlf, 2019-10-15:
We host the code of our `desklab` library on our own servers (https://git.desk-lab.de/). I hope this won't be an issue.

**Repository**:
https://git.desk-lab.de/desklab/desklab-arduino-lib

per1234, 2019-10-15:
Hi. Unfortunately we do not support repositories hosted on your own servers:

https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager
>Your library repository must be hosted on a major git-hosting website like GitHub, BitBucket or GitLab

Please either mirror the library on one of the hosting sites listed above. Please comment here once that is done with the new URL, or let us know if you want to cancel your request.

ax-sc, 2019-10-17:
The library has been migrated and is now hosted on GitHub.

**Repository:** [https://github.com/desklab/desklab-arduino-lib](https://github.com/desklab/desklab-arduino-lib)

jnsdrtlf, 2019-10-22:
I hope this did not cause any confusion, @ax-sc is part of our team. The link he shared is correct. I've been unavailable the last few days.

As @ax-sc said, we've migrated our code to GitHub and will continue to push updates to this repository ([desklab/desklab-arduino-lib](https://github.com/desklab/desklab-arduino-lib)). We understand your decision not to support self-hosted repositories.

ax-sc, 2019-10-24:
@per1234 If you´re still waiting for feedback we didn´t provide yet, please let us know.

cmaglie, 2019-10-25:
Done, sorry for the delay.

closed 2019-10-25 by cmaglie

jnsdrtlf, 2019-10-25:
No worries, thanks a lot!