arduino/Arduino #9323: Compiling Error

ArdinoFan, 2019-10-16:
Good day Sir/Mam,

I've used this code successfully  (flickering LED) before, but now the compiler gives me an compiling error after verification..., I will include a screen~capture, see it below. Please help.

Arduino-Fan,
Regards,
Schalk (from South Africa)

[image]

matthijskooijman, 2019-10-16:
It looks like you might have a 32bit vs 64bit problem or something similar, maybe you recently upgraded your OSX version? I've seen some issues regarding this recently.

In any case, you're running a fairly old version (1.6.12 is from three years ago), so please try to upgrade to the latest version. Very likely that will solve your problem.

ArdinoFan, 2019-10-16:
Good morning Sir/Mam,

Thanks for the speedy reply, I will try doing that and see if it works & then report back to you...! 😃

Christ-greetings,
✝️
Schalk Coetzer
 “I, your God, have a firm grip on you and I’m not letting go. I’m telling you, don’t panic. I’m right here to help you.”  (Isaiah 41:13)

ArdinoFan, 2019-10-17:
Good morning Sir/Mam,

As I’ve said yesterday after your answer on my question, I downloaded the new program and it immediately work. Thanks again for such speedy responce. 

Christ-greetings,
✝️
Schalk Coetzer
 “I, your God, have a firm grip on you and I’m not letting go. I’m telling you, don’t panic. I’m right here to help you.”  (Isaiah 41:13)