arduino/Arduino #9325: Correct code not working

Codemonk-adi, 2019-10-16:
I wrote the following code and the IDE is giving error Error compiling for board Arduino/Genuino Uno.
//code begins
int Right_Sensor=13;
int Central_Sensor=2;
int Left_Sensor=3;
int Motor_A_Enable=11;
int L1=4;
int L2=5;
int FlagL=0;
int FlagR=0;
int FlagC=0; 
int B1=6;
int B2=7;
int Motor_B_Enable=10;
void setup() {
  // put your setup code here, to run once:
  pinMode(Right_Sensor,INPUT);
  pinMode(Central_Sensor,INPUT);
  pinMode(Left_Sensor,INPUT);
  pinMode(L2,OUTPUT);
  pinMode(L1,OUTPUT);
  pinMode(B1,OUTPUT);
  pinMode(B2,OUTPUT);
  pinMode(10,OUTPUT);
  pinMode(11,OUTPUT);
 analogWrite(Motor_A_Enable,100);
 analogWrite(Motor_B_Enable,100);
}
void forward()
{ 
    digitalWrite(L1,HIGH);
    digitalWrite(L2,LOW);
    digitalWrite(B1,HIGH);
    digitalWrite(B2,LOW);
 }
void Stop()
{ 
    digitalWrite(L1,LOW);
    digitalWrite(L2,LOW);
    digitalWrite(B1,LOW);
    digitalWrite(B2,LOW);
 }
void right()
{ 
    digitalWrite(L1,HIGH);
    digitalWrite(L2,LOW);
    digitalWrite(B1,LOW);
    digitalWrite(B2,LOW);
 }
void left()
{ 
    digitalWrite(L1,LOW);
    digitalWrite(L2,LOW);
    digitalWrite(B1,HIGH);
    digitalWrite(B2,LOW);
 }
void loop() {
  // put your main code here, to run repeatedly:
FlagR=digitalRead(Right_Sensor);
FlagC=digitalRead(Central_Sensor);
FlagL=digitalRead(Left_Sensor);
if((FlagR==0)&&(FlagL==0))
{
forward();
}
if((FlagR==0)&&(FlagL==1))
{
  left();
}
if((FlagR==1)&&(FlagL==0))
{
right();
}
if((FlagR==1&&FlagL==1))
{
 forward();
}
}

matthijskooijman, 2019-10-16:
If there is an error, your code is likely not correct after all (though admittedly it looks ok at first glance). Also, your report is missing important details, such as at least the error message you get, what versions of things you are using and a good argument why that error message is not actually correct.

In any case, from what I can see now this seems more likely a problem with your code than a bug in the Arduino IDE, so this might not be the right place to get help (e.g. the Arduino forums might be a better place).

Feel free to add some more information if you're sure this is an Arduino / compiler bug, or feel free to close this issue and seek help with your code elsewhere.

closed 2019-10-16 by Codemonk-adi

PaulStoffregen, 2019-10-16:
https://www.arduino.cc/reference/en/language/variables/constants/integerconstants/