arduino/Arduino #9335: Dropdown menus appearing outside screen borders.

cls-02, 2019-10-20:
I am running the Official version of Arduino 1.8.10 under OpenSUSE Tumbleweed.
I have noticed that with nVidia graphics cards running in MetaMode under X11, large dropdown menus like the examples or libraries menu seem to be rendered partially outside of screen borders, resulting in unclickable menu items and unscrollable menus. This issue dissappears when MetaMode is disabled and a single display is used.
I'v attached a video of the issue I'm getting (zipped, GitHub doesn't like webms).
[Screencast from 10-20-2019 10:33:31 AM.zip](https://github.com/arduino/Arduino/files/3747656/Screencast.from.10-20-2019.10.33.31.AM.zip)

matthijskooijman, 2019-10-20:
Can you explain what MetaMode is? And what the movie shows exactly? I *think* it shows multiple screens stitched together and the long vertical menus extend slightly above the border of one screen into another screen, but can you explain a bit more clearly?

cls-02, 2019-10-20:
Metamode is the rather nasty way Nvidia does multiple monitors. It basically renders a big virtual screen that fits all of the screens and then outputs parts of it to respective displays. The white area in the movie is the unused area of the metamode screen. The movie shows that when there isn't enough space on the screen, Arduino simply puts menu items outside of the screen instead of correctly recognizing the screen border and scrolling correctly. The second part of the video shows that in fact, there are more menu items that don't show on the left screen.

gisforgirard, 2020-11-24:
> I am running the Official version of Arduino 1.8.10 under OpenSUSE Tumbleweed.
> I have noticed that with nVidia graphics cards running in MetaMode under X11, large dropdown menus like the examples or [...]

[simplescreenrecorder-2020-11-24_12.38.09.mp4.zip](https://github.com/arduino/Arduino/files/5591736/simplescreenrecorder-2020-11-24_12.38.09.mp4.zip)
this is how i found to get to the menu without having to turn everything off, for whatever reason this works...