arduino/Arduino #9337: Compiling Hangs When Connected to Wi-Fi Without Internet Access

TravelByRocket, 2019-10-21:
I am compiling for a HUZZAH32 that I have been updating over-the-air on a wifi network that is not connected to the internet. The compilation process reliably hangs on this step:

`/Users/myusername/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1/esptool --chip esp32 elf2image --flash_mode dio --flash_freq 80m --flash_size 4MB -o /var/folders/np/pb2s6nkx1vz6nny8gj1flrj80000gn/T/arduino_build_138615/longSiteLeft.ino.bin /var/folders/np/pb2s6nkx1vz6nny8gj1flrj80000gn/T/arduino_build_138615/longSiteLeft.ino.elf`

**Information to Reproduce:**

- Using Arduino IDE 1.8.10 and macOS 10.15
- Compiling for HUZZAH32 Feather

1. Connect to a wifi network that does not have internet access
2. Compilation hangs on the step pasted above
3. Switch to an internet-connected wifi network for the compilation to finish
4. Switch back to a non-connected wifi network for OTA update to upload and complete

matthijskooijman, 2019-10-21:
Thanks for your issue report.

However, if this is specific to the ESP an esptool, I think this is reported in the wrong place. This board is not supported by the Arduino IDE directly, but by a third-party core package you installed. The repository for that particular core is where you should be reporting this. I'm not entirely sure which core you're using, so you will have to see where to report exactly.

I'm going to proceed by closing this issue. Please re-report it in the right place. Thanks!

closed 2019-10-21 by matthijskooijman

TravelByRocket, 2019-10-21:
Thanks for the guidance! That makes sense and I will do that soon.