arduino/Arduino #9338: IDE Hangs on Startup Without Internet via WiFi

TravelByRocket, 2019-10-21:
If I start the Arduino IDE when I am connected to a wifi network that does not have internet access, it will give me the colored beach ball for quite some time. If I switch to a network with internet access or if I turn wifi off then it resolves immediately. I have disabled the option to check for updates on startup but that did not seem to make any difference. I have IDE version 1.8.10 and macOS 10.15.

matthijskooijman, 2019-10-21:
Can you start the IDE from the commandline? Maybe that gives some indication on what it's doing?

matthijskooijman, 2019-10-21:
Also, I've seen your other report (#9337) about delays related to the ESP HUZZAH board. Does the problem in this issue happen also when another board is selected, e.g. the Uno?

You say "for quite some time", so eventually the startup does complete?

TravelByRocket, 2019-10-21:
Command Line -- I'll work to pursue this suggestion and report back on that.

Hanging in My Other Report -- I had no problem when compiling for a HUZZAH8266 and WEMOS D1 Mini. This also supports your suggestion that I should report it to the ESP32 repo. 

Thanks for your questions and time!

TravelByRocket, 2019-10-24:
This issue happened at a remote site and I have not been back there since posting this issue. I tried to re-create the issue at home by disconnecting the ethernet cable from my router but I found that the Arduino IDE started as it should. The differences that come to mind are:
- The router at the remote site is a combined router+modem that was lying around, while at home I have a Google Wifi unit. Does this seem important in any way?
- I have ESP8266/32 boards running network ports for over-the-air updating so I wonder if the Arduino IDE is somehow getting stuck on looking at/for those on startup? I can still work on re-creating this at home.

TravelByRocket, 2019-10-28:
I did what I could to copy the remote site situation at home but instead the IDE worked as it should. I will try again this weekend at the remote site to open the IDE from the command line to try and catch more information about this hanging.