arduino/Arduino #9339: Files located under src/ are not found in 1.8.10

jonm111, 2019-10-21:
My sketch builds perfectly in 1.8.9.  As well as .ino, sketch uses source located under src/shared/

Upgrade to 1.8.10, sketch no longer builds.  Files included from e.g. "src/shared/blah.h" are no longer seen resulting in "No such file or directory" error at #include statement.

Revert to 1.8.9, everything is fine again.

facchinm, 2019-10-21:
Hi @jonm111 ,
can you provide a minimum verifiable sketch that exhibit the problem we can compile and analyze? Thanks

tobiasisenberg, 2019-12-17:
I noticed a similar issue. In 1.8.9 I can include external code with
`#include "external.h"`
provided that external.h and external.cpp are located in the same directory as the *.ino file. In 1.8.10 this does no longer work. I just tried to put together a simple example, yet interestingly my small generic example did not show the same effect. I tested some more and it turns out that the effect only happens for specific included file names, namely *.h and *.cpp files that start with RCS, see the attached example.
[MyOwnTest.zip](https://github.com/arduino/Arduino/files/3974002/MyOwnTest.zip)

Interestingly, if the name of *.ino file is "test" or "bug" I get a completely different compile error:
`panic: runtime error: invalid memory address or nil pointer dereference`
Very strange, you can try it out by renaming *.ino and directory to bug.

It would be great if this could be fixed.

matthijskooijman, 2019-12-17:
@tobiasisenberg, IIRC RCS is a version control system, so it might very well be that those files are excluded from the temporary sketch copy that is compiled in an attempt to exclude version control files (e.g. like `.git` or `.svn` directories). I suspect this is a completely different problem from the original poster, could you perhaps create a separate issue for this? The best place for this would, I think, be at https://github.com/arduino/arduino-cli which is where the relevant code now lives.

matthijskooijman, 2019-12-17:
@jonm111, we're still waiting for an example sketch that shows your problem. If you do not provide one, we cannot reproduce and fix this issue and will close it soon.

tobiasisenberg, 2019-12-17:
Interesting, thanks. What's interesting is that this worked fine in 1.8.9. Well, at least I have a work-around now for my own case, of course I am not sure about the case of @jonm111 --- whether this was the same effect or not.

tobiasisenberg, 2019-12-17:
Will create a separate issue.

per1234, 2020-08-04:
Since the information was never provided that would allow us to investigate this, and I have verified that `#include`s for files under the `src` subfolder of the sketch are working with the latest hourly build of the Arduino IDE, I'm going to close this.

@jonm111 if the issue still occurs for you with the latest hourly build of the Arduino IDE and you want to provide us with the details required to reproduce the issue, comment here and I'll be happy to reopen this.

closed 2020-08-04 by per1234