arduino/Arduino #9344: Arduino IDE 1.8.10 Compile Issue with Arduino/Genuino Micro

lloydroc, 2019-10-22:
Arduino IDE 1.8.10
Board Arduino/Genuino Micro - on back of board it says Board model R3.

This is especially troublesome since I get a different error compiling on the web editor.

When compiling anything I get the following error (note verbose for compilation is turned on):

```
Arduino: 1.8.10 (Mac OS X), Board: "Arduino/Genuino Micro"

/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/lloyd/Documents/Arduino/libraries -fqbn=arduino:avr:micro -vid-pid=0X2341_0X8037 -ide-version=10810 -build-path /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_build_683242 -warnings=none -build-cache /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_cache_994059 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_modified_sketch_891524/Blink.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/lloyd/Documents/Arduino/libraries -fqbn=arduino:avr:micro -vid-pid=0X2341_0X8037 -ide-version=10810 -build-path /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_build_683242 -warnings=none -build-cache /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_cache_994059 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.3.0.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino17.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-7.3.0-atmel3.6.1-arduino5.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_modified_sketch_891524/Blink.ino
Using board 'micro' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Detecting libraries used...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8037 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Micro\"" -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/micro /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_build_683242/sketch/Blink.ino.cpp -o /dev/null
Generating function prototypes...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega32u4 -DF_CPU=16000000L -DARDUINO=10810 -DARDUINO_AVR_MICRO -DARDUINO_ARCH_AVR -DUSB_VID=0x2341 -DUSB_PID=0x8037 "-DUSB_MANUFACTURER=\"Unknown\"" "-DUSB_PRODUCT=\"Arduino Micro\"" -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/micro /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_build_683242/sketch/Blink.ino.cpp -o /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_build_683242/preproc/ctags_target_for_gcc_minus_e.cpp
/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_build_683242/preproc/ctags_target_for_gcc_minus_e.cpp
panic: runtime error: index out of range

goroutine 1 [running]:
github.com/arduino/arduino-cli/legacy/builder/ctags.parseTag(0xc000324002, 0xa, 0xe)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/ctags/ctags_parser.go:203 +0xa22
github.com/arduino/arduino-cli/legacy/builder/ctags.(*CTagsParser).Parse(0xc0000d2060, 0xc000324000, 0x2ea, 0xc000198d90, 0xc000324000, 0x2ea, 0x600)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/ctags/ctags_parser.go:67 +0xf9
github.com/arduino/arduino-cli/legacy/builder.(*CTagsRunner).Run(0x1bacff0, 0xc000071000, 0x1bacff0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/ctags_runner.go:71 +0x4bd
github.com/arduino/arduino-cli/legacy/builder.(*ContainerAddPrototypes).Run(0x1bacff0, 0xc000071000, 0x1bacff0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/container_add_prototypes.go:65 +0x4b8
github.com/arduino/arduino-cli/legacy/builder.runCommands(0xc000071000, 0xc0000a9fb0, 0x1, 0x1, 0x1, 0xc0000a9fb0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xdf
github.com/arduino/arduino-cli/legacy/builder.(*PreprocessSketch).Run(0x1bacff0, 0xc000071000, 0x1bacff0, 0x0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:146 +0xc7
github.com/arduino/arduino-cli/legacy/builder.runCommands(0xc000071000, 0xc000527c20, 0x20, 0x20, 0x1, 0xc0000b9ab8, 0x9)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:210 +0xdf
github.com/arduino/arduino-cli/legacy/builder.(*Builder).Run(0xc0000b9e70, 0xc000071000, 0x1bacef0, 0xc0000a97a0)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:117 +0xd28
github.com/arduino/arduino-cli/legacy/builder.RunBuilder(...)
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-cli/legacy/builder/builder.go:226
main.main()
	/home/jenkins/workspace/arduino-builder-all-cross-cli-inception/src/github.com/arduino/arduino-builder/main.go:398 +0x800
/Applications/Arduino.app/Contents/Java/arduino-builder returned 2
Error compiling for board Arduino/Genuino Micro.
```

facchinm, 2019-10-23:
Hi @lloydroc ,
which version of osx are you running? It looks like ctags is crashing or reporting strage output.
Would you mind trying to execute `/Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_build_683242/preproc/ctags_target_for_gcc_minus_e.cpp` (of course changing `/var/folders/z8/cgzdm8hn0...`  into an existing folder) and report the output here? Thanks!

lloydroc, 2019-10-23:
Hello @facchinm here is the output:

```
$ /Applications/Arduino.app/Contents/Java/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_build_683242/preproc/ctags_target_for_gcc_minus_e.cpp
  Option: -R
Considering option file ./.ctags: not found
Reading initial options from command line
  Option: -u
  Option: --language-force=c++
  Option: -f -
  Option: --c++-kinds=svpf
  Option: --fields=KSTtzns
  Option: --line-directives=
Reading command line arguments
OPENING /var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_build_683242/preproc/ctags_target_for_gcc_minus_e.cpp as C++ language file
setup	/var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_modified_sketch_305294/Blink.ino	/^void setup() {$/;"	kind:function	line:26	signature:()	returntype:void
loop	/var/folders/z8/cgzdm8hn0kn09shs2kcn46xr0000gn/T/arduino_modified_sketch_305294/Blink.ino	/^void loop() {$/;"	kind:function	line:32	signature:()	returntype:void
$
```

I am running macOS Catalina version 10.15.

I do use ctags and have a `.ctags` file.

```
$ cat ~/.ctags
--exclude=*.class
--exclude=*.html
--exclude=*.jar
--exclude=*.erl
--exclude=*.hrl
--exclude=*.fxs
-V
-R
$
```

If I comment it out the lines in the `.ctags` it works and that was the problem. Thanks! Maybe there is a way for the Arudino code to ignore this file for users who use it elsewhere? I surely didn't want it to kick in when building Arduino code.

closed 2019-10-23 by lloydroc

facchinm, 2019-10-24:
Good to hear you solved! Personally it's the first time I discover that `~/.ctags` is a thing; if there's a flag to exclude it we should absolutely add to our commandline

zapgods, 2019-11-24:
I tried rename folder library >blynk-library-master >>blynk-library. it woked.

lloydroc, 2019-11-24:
I looked and I don't see a good way to override if people have their own
ctags files. Best could be to maybe throw warning as the install could look
before.

oqibidipo, 2019-12-22:
`--options=NONE` as the first command line option disables automatic reading  of configuration files.

http://ctags.sourceforge.net/ctags.html#OPTIONS

muekno, 2019-12-26:
similar problem, compiling a simple test sketch for DHT22 on arduino nano. First time I got that problem. All my other wide more complex sketches (with no DHT22) but other peripherals like CAN board, real time clock etc. are compiling well and running. 
Need hint or help.
Thanks Rainer
Merry Christmas

muekno, 2019-12-26:
Solved, found two DHT Libraries, delete the older one, can compile now

lloydroc, 2019-12-27:
That would be great to have as an option for the Arduino build ... This way
we won't have the Arduino build using people's $HOMe/.ctags and getting
strange issues. Or maybe throw a warning if it's detected?

/Lloyd