arduino/Arduino #9345: SSD1306 Examples fail on Arduino Nano 33 BLE

matthewsellis, 2019-10-22:
Having tried both the SSD1306 128x64 SPI and I2C examples from the Adafruit SSD1306 library with the Board set to 'Arduino nano 33 BLE', I see the following Errors when compiling:

`C:\Users\matth\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp: In member function 'boolean Adafruit_SSD1306::begin(uint8_t, uint8_t, boolean, boolean)':

C:\Users\matth\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp:482:28: error: 'portOutputRegister' was not declared in this scope

 dcPort    = (PortReg *)portOutputRegister(digitalPinToPort(dcPin));

                        ^~~~~~~~~~~~~~~~~~
C:\Users\matth\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp:483:17: error: 'digitalPinToBitMask' was not declared in this scope

 dcPinMask = digitalPinToBitMask(dcPin);

             ^~~~~~~~~~~~~~~~~~~
C:\Users\matth\Documents\Arduino\libraries\Adafruit_SSD1306\Adafruit_SSD1306.cpp:483:17: note: suggested alternative: 'digitalPinToPinName'

 dcPinMask = digitalPinToBitMask(dcPin);

             ^~~~~~~~~~~~~~~~~~~

             digitalPinToPinName
Multiple libraries were found for "SPI.h"
Used: C:\Users\matth\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.2\libraries\SPI
Multiple libraries were found for "Wire.h"
Used: C:\Users\matth\AppData\Local\Arduino15\packages\arduino\hardware\mbed\1.1.2\libraries\Wire
Multiple libraries were found for "Adafruit_GFX.h"
Used: C:\Users\matth\Documents\Arduino\libraries\Adafruit_GFX_Library
Multiple libraries were found for "Adafruit_SSD1306.h"
Used: C:\Users\matth\Documents\Arduino\libraries\Adafruit_SSD1306
exit status 1
Error compiling for board Arduino Nano 33 BLE.
`

I get a clean compile with no errors reported if I switch to (e.g.) original Arduino Nano board.

I originally raised this as a defect on the Adafruit ssd1306 library, and received a reply from Ladyada to the effect the issue is missing standard functions in the nRF528x library [link](https://github.com/adafruit/Adafruit_SSD1306/issues/150)

Package versions used:

- Arduino nRF528x Boards (Mbed OS) by Arduino version 1.1.2
- Adafruit GFX Library 1.6.1
- Adafruit SSD1306 Library 1.3.0
- Arduino IDE 1.8.10 on Win 10 1903

As an aside, some of the errors above relate to multiple versions of both Adafruit libraries being installed, but when I delete both libraries from the path specified in the errors above, the compile fails, reporting libraries missing. The library duplicate errors do not show when Arduino nano (non 33 BLE variant) is selected.

facchinm, 2019-10-23:
Hi @matthewsellis ,
the issue is indeed in Adafruit library; this line https://github.com/adafruit/Adafruit_SSD1306/blob/master/Adafruit_SSD1306.h#L51 adds `HAVE_PORTREG` define to any ARM board which is wrong.
I'll submit a patch there, closing here.

closed 2019-10-23 by facchinm

amaussa, 2021-04-18:
see https://forum.arduino.cc/t/arduino-nano-ble-33-sense-oled-ssd1327-128x128-spi/636461