arduino/Arduino #9348: IDE crashes at startup when using jre-openjdk 13

matt24smith, 2019-10-23:
Arduino IDE crashes at startup when using jre-openjdk-13. I received the following error message: 
```
Picked up JAVA_TOOL_OPTIONS: 
java.lang.ExceptionInInitializerError
	at processing.app.helpers.PreferencesMap.load(PreferencesMap.java:100)
	at processing.app.helpers.PreferencesMap.load(PreferencesMap.java:74)
	at processing.app.PreferencesData.init(PreferencesData.java:56)
	at processing.app.BaseNoGui.initParameters(BaseNoGui.java:835)
	at processing.app.Base.<init>(Base.java:213)
	at processing.app.Base.main(Base.java:151)
Caused by: java.lang.StringIndexOutOfBoundsException: begin 0, end 3, length 2
	at java.base/java.lang.String.checkBoundsBeginEnd(String.java:3720)
	at java.base/java.lang.String.substring(String.java:1909)
	at processing.app.legacy.PApplet.<clinit>(PApplet.java:38)
	... 6 more
```

The problem occurs with IDE versions 1.8.10 and 1.8.9. When using downgraded jre-openjdk version 10 the IDE can run, but with the following warnings. Maybe a GTK issue? My operating system is linux 5.3.7. 

```
Picked up JAVA_TOOL_OPTIONS: 
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by processing.app.linux.GTKLookAndFeelFixer (file:/usr/share/arduino/lib/arduino-core.jar) to field com.sun.java.swing.plaf.gtk.GTKLookAndFeel.styleFactory
WARNING: Please consider reporting this to the maintainers of processing.app.linux.GTKLookAndFeelFixer
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```

Gspin96, 2019-10-31:
On arch linux, this issue was fixed after installing the latest jre-openjdk-13 package (13.0.1.u9-1)

matt24smith, 2019-10-31:
Can confirm, appears to be working now

closed 2019-10-31 by matt24smith