arduino/Arduino #9349: [Library Manager] Please add GroveDriverPack library

matsujirushi, 2019-10-24:
Hi,

I created the library.
Could you add this to a list of libraries?
https://github.com/SeeedJP/GroveDriverPack

Thanks.

per1234, 2019-10-25:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

matsujirushi, 2019-10-25:
Oh, sorry.
created.
https://github.com/SeeedJP/GroveDriverPack/releases/tag/1.4.0

cmaglie, 2019-10-31:
Done.

closed 2019-10-31 by cmaglie