arduino/Arduino #9351: arm-none-eabi-g++: no such file or directory

1cookspe, 2019-10-25:
I am trying to upload the "Blink" example to my MKR 1000, however I keep getting the same error: "fork/exec /Users/username/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/7-2017q4/bin/arm-none-eabi-g++: no such file or directory". I have installed the SAMD board from the Board Manager in the IDE, and I also updated GCC-C++ on my machine. I am using a Mac on the Mac Arduino IDE. I have also tried uninstalling and reinstalling the Arduino IDE. I tested my MKR 1000 on another computer and it worked perfectly. Any ideas how I can fix this?

facchinm, 2019-10-25:
You probably need to reinstall the SAMD package from board manager since the compiler didn't extract properly. In the worst case you can delete `/Users/username/Library/Arduino15/packages/arduino` folder and then reinstall the core

MAKIO135, 2020-01-11:
Same problem here on Mac with the SAMD board package from the Board Manager.
@facchinm What do you mean by reinstall the core?

luisfigo27, 2023-06-16:
I had the same problem, but using a different board, i used teensyduino, when I followed the instructions to enable teensy in Arduino IDE. "fork/exec /arm/bin/arm-none-eabi-g++: no such file or directory " these error occurs. Do anybody had a solution ? The purpose of my work is to impliment micro-ros. I had uninstall and reinstall the arduino IDE, at first an undefined reference error, and then I reinstall the micro-ros libraries and then the mentioned error occured. 
[image]

per1234, 2023-06-16:
Hi @luisfigo27. Please update to Arduino IDE 2.x. The download links are listed here:

https://www.arduino.cc/en/software

After that, install the **Teensy** boards platform by following these instructions:

https://www.pjrc.com/arduino-ide-2-0-0-teensy-support/

That should solve the error. If it still occurs, post the updated description of the problem on the Arduino Forum and we'll help you to fix the error over there:

https://forum.arduino.cc/

closed 2023-06-16 by per1234