arduino/Arduino #9354: [Help] How to compile library for creating Excel XLSX files?

rtek1000, 2019-10-26:
Hello, I found this 'C' library which promises to be very useful for generating files in datalogger projects, unfortunately CSV files need to be converted manually, it would be great to be able to write xlsx files directly.

After merging all the files into one folder and editing the headers to work with the new location, the compiler returned a very unremarkable error on the internet:

> 
> c:/arduino ide/arduino-1.8.10-windows-portable/portable/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7-m\libc.a(lib_a-openr.o): In function `_open_r':
> 
> openr.c:(.text._open_r+0x10): undefined reference to `_open'
> 
> c:/arduino ide/arduino-1.8.10-windows-portable/portable/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/../lib/gcc/arm-none-eabi/4.8.3/../../../../arm-none-eabi/lib/armv7-m\libc.a(lib_a-unlinkr.o): In function `_unlink_r':
> 
> unlinkr.c:(.text._unlink_r+0xc): undefined reference to `_unlink'
> 
> collect2.exe: error: ld returned 1 exit status

I found this reference:
> The GCC toolchain is only half of what you need to create a working executable*. The other half is the runtime library. The runtime includes crt0.o, which contains the entry point (the code that calls main()), and generally a libc that contains the standard C functions (strcmp(), memcpy(), etc) as well as the system calls (open(), read(), and others). You need to find a source for these. If you're targeting an embedded Linux or BSD machine, you'll have to find out what libc your target is using. It's probably either GNU libc, BSD libc, newlib, or uclibc. You can get these and build them yourself, or they may be available already with your OS.
> 
> *unless you're building a freestanding binary, but this doesn't look to be what you're doing.
> 
Source: https://stackoverflow.com/questions/6493122/missing-something-in-arm-g

So I would like to know, from anyone who understands a little more about the build that produces the Arduino IDE, if it is possible to compile this library to work with Arduino DUE.

Thank you.

Ref.:
> Libxlsxwriter is a C library that can be used to write text, numbers, formulas and hyperlinks to multiple worksheets in an Excel 2007+ XLSX file.
> 
> It supports features such as:
> 
>     100% compatible Excel XLSX files.
>     Full Excel formatting.
>     Merged cells.
>     Defined names.
>     Autofilters.
>     Charts.
>     Data validation and drop down lists.
>     Worksheet PNG/JPEG images.
>     Support for adding Macros.
>     Memory optimization mode for writing large files.
>     Source code available on GitHub.
>     FreeBSD license.
>     ANSI C.
>     Works with GCC, Clang, Xcode, MSVC 2015, ICC, TCC, MinGW, MingGW-w64/32.
>     Works on Linux, FreeBSD, OpenBSD, OS X, iOS and Windows. Also works on MSYS/MSYS2 and Cygwin.
>     Compiles for 32 and 64 bit.
>     Compiles and works on big and little endian systems.
>     The only dependency is on zlib.

https://github.com/jmcnamara/libxlsxwriter

Zlib data compression library:
https://github.com/madler/zlib

[Arduino_XLSX_test.zip](https://github.com/arduino/Arduino/files/3774402/Arduino_XLSX_test.zip)

cmaglie, 2019-10-30:
Please use the forums for help on your project:

https://forum.arduino.cc/

closed 2019-10-30 by cmaglie

rtek1000, 2019-10-30:
https://forum.arduino.cc/index.php?topic=642853.0