arduino/Arduino #9364: [Library Manager] Change name for esp8266-oled-ssd1306

marcelstoer, 2019-10-28:
https://github.com/ThingPulse/esp8266-oled-ssd1306 changed the name from 

"ESP8266 and ESP32 Oled Driver for SSD1306 display"

to

"ESP8266, ESP32 and Mbed-OS OLED Driver for SSD1306 displays"

with the latest release v4.1 as per https://github.com/ThingPulse/esp8266-oled-ssd1306/commit/f751540a4f420bdf0afcb179329f07fdc3d1a805#diff-ea32f5b85735296fd64b577e91a19607. Thanks for adjusting so that the indexer picks this up.

marcelstoer, 2019-10-31:
@per1234 Thanks for tagging this correctly. Do you have an ETA for the renaming?

cmaglie, 2019-10-31:
@marcelstoer the comma `,` is not permitted in the library name, you must remove it from the name otherwise we can't change the name.

Also, wouldn't be better to change the name to something more... architecture independent? :-)
Something like "ThingPulse SSD1306" for example. You can still keep the list of compatible MCU in the description.

marcelstoer, 2019-11-01:
Thank you @cmaglie, noted. 

> change the name to something more... architecture independent?

Yep, I was thinking about that, too.

After I change the name do I have to create a new release/tag or will the indexer script look at the name on the `master` branch?

marcelstoer, 2019-11-04:
@per1234 can you maybe help out answering the above question?

cmaglie, 2019-11-04:
you must make a new release tag

marcelstoer, 2019-11-10:
Alright, for various reasons I opted for "ESP8266 and ESP32 OLED driver for SSD1306 displays". Please update it in your index. The latest release 4.1.0 is properly tagged.

cmaglie, 2019-11-11:
Done.

closed 2019-11-11 by cmaglie