arduino/Arduino #9365: Serial plotter,Do a display range adjustable X-axis，Add repeatable options when displaying the same contents.

breaker123, 2019-10-28:
Please read the Issues section of the Contributing Rules at the "Contributing" link to the right before submitting an issue report.
Is your feature request related to a problem? Please describe.
Hi, Awesome work! well done.I have used Arduino for 2 years,I have made many interesting projects.I wanna thank your team with fantastic work.
But ,I also met questions, When I use the serial plotter, I want to extend the display range of the horizontal axis time so that the latest data acquisition and the previous data acquisition can be displayed on the same graph.Second? when the x-coordinate data is a cyclic change in a range, I want the x-coordinate data to be displayed within the range instead of forward.
thanks for your reading.i look forward to your response. my email address is 1701213930@pku.edu.cn , feel free to contact.

Describe the solution you'd like
Do a display range adjustable X-axis，Add repeatable options when displaying the same contents.

EvertSchut, 2020-01-23:
Hi,
The plotter is indeed handy. What could help me now a lot:
To be able to fix the scales
Freeze the picture (e.g. to make a screencap)
Save output for later analysis.
If it would be added, that would be very nice.
I saw some posts of people doing this in the source code, but it is not so easy for me to compile it and add/replace this part in the Arduino package. Would be nice if it was an option or add in.
Thanks for the work!

madsdyd, 2020-04-07:
Your issue request / wish is really two issues in one:

> When I use the serial plotter, I want to extend the display range of the horizontal axis time so that the latest data acquisition and the previous data acquisition can be displayed on the same graph.

This is adressed in #7454. In essence this feature request is a duplicate of #7454. I suggest you go to this request and add a comment that you would like this feature.

> Second? when the x-coordinate data is a cyclic change in a range, I want the x-coordinate data to be displayed within the range instead of forward.

The way the serial monitor is designed, there is no x coordinate: Every datapoint is just appended at the end of the existing datapoints. After 500 datapoints appended, everytime a new datapoint is appended, the oldests is removed. Changing this approach would require both a redesign of the existing Serial Plotter, and breaking "compatibility" with all existing Arduino code that uses it.

@per1234 I suggest you close this as a duplicate of #7454

breaker123, 2020-04-07:
thanks your response, but how can I get the code of Arduino and fix it?

madsdyd, 2020-04-07:
> thanks your response, but how can I get the code of Arduino and fix it?

If you want to build the Arduino codebase from scratch, visit this page: https://github.com/arduino/Arduino/wiki/Building-Arduino and start from there. However, this requires knowledge of Java, etc.

What I really think you want is to express your interest for getting #7454 merged into mainline, and released. The best way to do this, IMHO, is to visit this issued and express your interest in getting it resolved.

breaker123, 2020-04-07:
thanks I really appreaciate

  

  
 
 
&mdash;

madsdyd, 2020-04-07:
@EvertSchut 

> To be able to fix the scales

The only scale that makes sense to fix, is the y scale. This is really a new feature request. Please add a new issue, and label it (if possible) with "SerialPlotter" and "feature request" labels.

> Freeze the picture (e.g. to make a screencap)
Also really a new feature request. See above.

> Save output for later analysis.
And a third feature request. See above.

In summary: Please add seperate issues for all of the above.

breaker123, 2020-04-07:
Ok ,thanks

madsdyd, 2020-04-07:
> @EvertSchut
> 
> > To be able to fix the scales
> 
> The only scale that makes sense to fix, is the y scale. This is really a new feature request. Please add a new issue, and label it (if possible) with "SerialPlotter" and "feature request" labels.

This feature request already exist: #4527  (Do not create a new one).
> 
> > Freeze the picture (e.g. to make a screencap)
> 
> Also really a new feature request. See above.

This feature request already exist: #4860   (Do not create a new one).

per1234, 2020-04-14:
>@per1234 I suggest you close this as a duplicate of #7454

Thanks for your suggestion, but we generally only close issues fixed by a PR once that PR has been merged.

cry0g3n-git, 2020-04-14:
Watching. Thanks