arduino/Arduino #9366: dot_a_linkage | error: libraries\dotATest\dotATest.a: No such file or directory

aliebold, 2019-10-28:
I am attempting to use the dot_a_linkage feature in a simple test library and the compiler is throwing an error saying that it can not find the .a file exactly where it is located and named. I have tried putting the .a file in the src folder and the root (and both...).

Here is the library:

dotATest.h
```
#ifndef DOTATEST_H
#define DOTATEST_H

int multiplyBy2(int x);

#endif /* End DOTATEST_H */
```

dotATest.cpp
```
#include "dotATest.h"

int multiplyBy2(int x) {
 return 2*x;
} 
```
I compile the .cpp file with the command:

`g++ -c dotATest.cpp -o dotATest.o`

Then I archive the .o file with this command:

`ar crf dotATest.a dotATest.o`

Then I create a v1.5 library structure with dotATest.h and dotATest.a in the src folder and a library properties file in the root directory.

library.properties
```
name=dotATest
version=1.0.0
author=Aaron Liebold
maintainer=Aaron Liebold
sentence=A simple library to test the dot_a_linkage
paragraph=
category=Data Processing
dot_a_linkage=true
url=
architectures=*
include=dotATest.h
```

I am testing the use of the library with the following sketch:
```
#include <dotATest.h>

void setup() {
  // put your setup code here, to run once:
  Serial.begin(2000000);
}

void loop() {
  if (millis()%1000 == 0) {
    Serial.print(millis() % 1000);
    Serial.print(" x 2 = ");
    Serial.print(multiplyBy2((int) millis() % 1000));
    Serial.println();
  }
}
```
And when it compiles I get the following error:
```
Arduino: 1.8.10 (Windows 10), Board: "Arduino/Genuino Uno"

...

Using previously compiled file: C:\Users\Aaron\AppData\Local\Temp\arduino_build_884537\core\core.a
Linking everything together...
"C:\\Program Files (x86)\\Arduino\\hardware\\tools\\avr/bin/avr-gcc" -Os -g -flto -fuse-linker-plugin -Wl,--gc-sections -mmcu=atmega328p -o "C:\\Users\\Aaron\\AppData\\Local\\Temp\\arduino_build_884537/dotOTestSketch.ino.elf" "C:\\Users\\Aaron\\AppData\\Local\\Temp\\arduino_build_884537\\sketch\\dotOTestSketch.ino.cpp.o" "C:\\Users\\Aaron\\AppData\\Local\\Temp\\arduino_build_884537\\libraries\\dotATest\\dotATest.a" "C:\\Users\\Aaron\\AppData\\Local\\Temp\\arduino_build_884537/core\\core.a" "-LC:\\Users\\Aaron\\AppData\\Local\\Temp\\arduino_build_884537" -lm
avr-gcc: error: C:\Users\Aaron\AppData\Local\Temp\arduino_build_884537\libraries\dotATest\dotATest.a: No such file or directory

Multiple libraries were found for "dotATest.h"
 Used: C:\Working_Files_LOCAL\Arduino\libraries\dotATest
Using library dotATest at version 1.0.0 in folder: C:\Working_Files_LOCAL\Arduino\libraries\dotATest 
exit status 1
Error compiling for board Arduino/Genuino Uno.
```

I checked the AppData\Local\Temp\arduino_build_884537\libraries\dotATest folder and it was empty. It seems that dotATest.a is not being moved into the temp directory for the compilation. Is something not set correction in my library.properties? Any insights would be helpful.

Thanks,
Aaron

facchinm, 2019-10-29:
Hi Aaron,
you are making a little confusion about `dot_a_linkage` and `precompiled` libraries.
DotALinkage is needed to avoid double declaration on some very particular libraries (declaring interrupts handlers, for example)

What you need is `precompiled=true` field; to generate the .a, though, you need to use the right compiler (`avr-gcc` in this case) and copy the .a in `src/$build.mcu` folder

See https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format for the `precompiled` field format

closed 2019-10-29 by facchinm

matthijskooijman, 2019-10-29:
> DotALinkage is needed to avoid double declaration on some very particular libraries (declaring interrupts handlers, for example)

Actually, I think it is mostly useful for allowing code to be optimized away that involves interrupt handlers (which are normally never optimized away). Just a small nitpick, I think the rest of your comment applies :-)