arduino/Arduino #9372: Global platform.txt

fabiuz7, 2019-10-29:
Hi all,
I was using the global **platform.txt**, (the one stored at path/to/your/app/hardware/platform.txt) to execute some custom prebuild hooks. I was working until I updated to Arduino IDE version 1.8.10. I switched back to version 1.8.9 and the hooks work as usual. Is it changed something in Arduino toolchain management?

facchinm, 2019-10-29:
@cmaglie I think this is related with the switch of arduino-builder to CLI
@fabiuz7 can you provide here the exact setting you were overriding and the expected result? Thanks!

fabiuz7, 2019-10-29:
Actually I used that file for the following configuration:

> recipe.hooks.sketch.prebuild.1.pattern=git-version.bat {build.source.path}/src {build.path}/sketch/src

Which is a small script to embed GIT tag in source code. [More detail](https://github.com/fabiuz7/git-describe-arduino)

I can observe that this script is never run.