arduino/Arduino #9376: [Library Manager] Please add avr_stl

ciband, 2019-10-30:
Please add my library "avr_stl" to the library manager.

https://github.com/ciband/avr_stl.git

per1234, 2019-10-31:
Hi. The `name` value in library.properties for your most recent release is ArduinoSTL. That name is already taken (as well as being prohibited for use in new Library Manager additions due to starting with the word "arduino"). I see you have since changed the name value to avr_stl, which is fine, but you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that contains the change to the new `name` value.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

ciband, 2019-10-31:
@per1234  I have created a v1.2.0 release as you have requested.  I think the repo is ready.  Please let me know if you need anything else. 

Thank you.

cmaglie, 2019-10-31:
Done.

closed 2019-10-31 by cmaglie