arduino/Arduino #9379: code complete not work at all in linux beta 64bit 1.9.0 version

alirezaimi, 2019-10-31:
Hi
the code completion not work atall for me in debian linux 64b version of arduino ide beta (1.9.0).

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading hardware from: /home/ali/arduino-PR-beta1.9-BUILD-116/hardware"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading package arduino from: /home/ali/arduino-PR-beta1.9-BUILD-116/hardware/arduino"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded platform" platform="arduino:avr@1.8.1"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading package discoverers from: /home/ali/arduino-PR-beta1.9-BUILD-116/hardware/discoverers"

    time="2019-10-31T11:30:06+03:30" level=warning msg="Can't determine bundle platform version for discoverers:disco, no valid release found"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded platform" platform="discoverers:disco@"

    time="2019-10-31T11:30:06+03:30" level=info msg="Excluding directory: /home/ali/arduino-PR-beta1.9-BUILD-116/hardware/tools"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading hardware from: /home/ali/.arduino15/packages"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading package builtin from: /home/ali/.arduino15/packages/builtin"

    time="2019-10-31T11:30:06+03:30" level=info msg="Checking existence of 'tools' path: /home/ali/.arduino15/packages/builtin/tools"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading tools from dir: /home/ali/.arduino15/packages/builtin/tools"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool="builtin:ctags@5.8-arduino11"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool="builtin:serial-discovery@1.0.0"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading package esp8266 from: /home/ali/.arduino15/packages/esp8266/hardware"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded platform" platform="esp8266:esp8266@2.5.2"

    time="2019-10-31T11:30:06+03:30" level=info msg="Checking existence of 'tools' path: /home/ali/.arduino15/packages/esp8266/tools"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading tools from dir: /home/ali/.arduino15/packages/esp8266/tools"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool="esp8266:mkspiffs@2.5.0-3-20ed2b9"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool="esp8266:python@3.7.2-post1"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool="esp8266:xtensa-lx106-elf-gcc@2.5.0-3-20ed2b9"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading tools from bundle dir: /home/ali/arduino-PR-beta1.9-BUILD-116/tools-builder"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading tools from dir: /home/ali/arduino-PR-beta1.9-BUILD-116/tools-builder"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool=":arduino-preprocessor@0.1.5"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool=":ctags@5.8-arduino11"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading tools from bundle dir: /home/ali/arduino-PR-beta1.9-BUILD-116/hardware/tools/avr"

    time="2019-10-31T11:30:06+03:30" level=info msg="Found builtin_tools_versions.txt"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool="arduino:avrdude@6.3.0-arduino17"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool="arduino:arduinoOTA@1.3.0"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool="arduino:avr-gcc@7.3.0-atmel3.6.1-arduino5"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading tools from bundle dir: /home/ali/.arduino15/packages"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loading tools from dir: /home/ali/.arduino15/packages"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool=":builtin@tools"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool=":esp8266@hardware"

    time="2019-10-31T11:30:06+03:30" level=info msg="Loaded tool" tool=":esp8266@tools"
[... 53 lines cut ...]
	    at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1103)
	    at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:974)
	    at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:800)
	    at java.awt.Component.dispatchEventImpl(Component.java:4760)
	    at java.awt.Container.dispatchEventImpl(Container.java:2297)
	    at java.awt.Window.dispatchEventImpl(Window.java:2746)
	    at java.awt.Component.dispatchEvent(Component.java:4711)
	    at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	    at java.awt.EventQueue.access$500(EventQueue.java:97)
	    at java.awt.EventQueue$3.run(EventQueue.java:709)
	    at java.awt.EventQueue$3.run(EventQueue.java:703)
	    at java.security.AccessController.doPrivileged(Native Method)
	    at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	    at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
	    at java.awt.EventQueue$4.run(EventQueue.java:733)
	    at java.awt.EventQueue$4.run(EventQueue.java:731)
	    at java.security.AccessController.doPrivileged(Native Method)
	    at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
	    at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	    at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	    at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	    at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	    at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	    at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	    at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

alirezaimi, 2019-11-02:
I remove old `.arduino` dir and reopen it again and install esp8266 lib again, but problem still exist :

	com.fasterxml.jackson.databind.exc.MismatchedInputException: No content to map due to end-of-input
	 at [Source: (String)""; line: 1, column: 0]
		at com.fasterxml.jackson.databind.exc.MismatchedInputException.from(MismatchedInputException.java:59)
		at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:4133)
		at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:3988)
		at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2992)
		at cc.arduino.autocomplete.ClangCompletionProvider.getCompletionsImpl(ClangCompletionProvider.java:99)
		at org.fife.ui.autocomplete.CompletionProviderBase.getCompletions(CompletionProviderBase.java:104)
		at org.fife.ui.autocomplete.AutoCompletion.refreshPopupWindow(AutoCompletion.java:801)
		at org.fife.ui.autocomplete.AutoCompletion$AutoCompleteAction.actionPerformed(AutoCompletion.java:1351)
		at javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1668)
		at javax.swing.JComponent.processKeyBinding(JComponent.java:2882)
		at javax.swing.JComponent.processKeyBindings(JComponent.java:2929)
		at javax.swing.JComponent.processKeyEvent(JComponent.java:2845)
		at java.awt.Component.processEvent(Component.java:6316)
		at java.awt.Container.processEvent(Container.java:2239)
		at java.awt.Component.dispatchEventImpl(Component.java:4889)
		at java.awt.Container.dispatchEventImpl(Container.java:2297)
		at java.awt.Component.dispatchEvent(Component.java:4711)
		at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
		at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:835)
		at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1103)
		at java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:974)
		at java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:800)
		at java.awt.Component.dispatchEventImpl(Component.java:4760)
		at java.awt.Container.dispatchEventImpl(Container.java:2297)
		at java.awt.Window.dispatchEventImpl(Window.java:2746)
		at java.awt.Component.dispatchEvent(Component.java:4711)
		at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
		at java.awt.EventQueue.access$500(EventQueue.java:97)
		at java.awt.EventQueue$3.run(EventQueue.java:709)
		at java.awt.EventQueue$3.run(EventQueue.java:703)
		at java.security.AccessController.doPrivileged(Native Method)
		at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
		at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:84)
		at java.awt.EventQueue$4.run(EventQueue.java:733)
		at java.awt.EventQueue$4.run(EventQueue.java:731)
		at java.security.AccessController.doPrivileged(Native Method)
		at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:74)
		at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
		at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
		at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
		at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
		at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
		at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
		at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
	com.fasterxml.jackson.databind.exc.MismatchedInputException: No content to map due to end-of-input
	 at [Source: (String)""; line: 1, column: 0]
		at com.fasterxml.jackson.databind.exc.MismatchedInputException.from(MismatchedInputException.java:59)
		at com.fasterxml.jackson.databind.ObjectMapper._initForReading(ObjectMapper.java:4133)
		at com.fasterxml.jackson.databind.ObjectMapper._readMapAndClose(ObjectMapper.java:3988)
		at com.fasterxml.jackson.databind.ObjectMapper.readValue(ObjectMapper.java:2992)
		at cc.arduino.autocomplete.ClangCompletionProvider.getCompletionsImpl(ClangCompletionProvider.java:99)
		at org.fife.ui.autocomplete.CompletionProviderBase.getCompletions(CompletionProviderBase.java:104)
		at org.fife.ui.autocomplete.AutoCompletion.refreshPopupWindow(AutoCompletion.java:801)
		at org.fife.ui.autocomplete.AutoCompletion$AutoCompleteAction.actionPerformed(AutoCompletion.java:1351)
		at javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1668)
		at javax.swing.JComponent.processKeyBinding(JComponent.java:2882)
		at javax.swing.JComponent.processKeyBindings(JComponent.java:2929)
		at javax.swing.JComponent.processKeyEvent(JComponent.java:2845)
		at java.awt.Component.processEvent(Component.java:6316)
		at java.awt.Container.processEvent(Container.java:2239)
		at java.awt.Component.dispatchEventImpl(Component.java:4889)
		at java.awt.Container.dispatchEventImpl(Container.java:2297)
		at java.awt.Component.dispatchEvent(Component.java:4711)
		at java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1954)
		at java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:835)
		at java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1103)
[... 45 lines cut ...]
[... cut ...]
		at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
		at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
		at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
		at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
		at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
		at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
		at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

facchinm, 2019-11-04:
Autocomplete on esp8266 has never been officially supported (due to clang vs gcc being used to compile). I'd leave the issue open but it won't likely be solved in the short period