arduino/Arduino #9386: Error opening serial port and can't upload sketch

dancind, 2019-11-01:
Hello, I am attempting to do my first project and I seem to already be running into problems. I was following a tutorial and was told to click the Serial Monitor button. When I do so, I receive the following error: 

> Arduino: 1.8.10 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
> 
> Error opening serial port 'COM3'. (Port busy)

I decided to truck ahead and try to upload a sketch and got a much more substantial error. Here is what that one said:

> Arduino: 1.8.10 (Windows 10), Board: "Arduino/Genuino Mega or Mega 2560, ATmega2560 (Mega 2560)"
> 
> Sketch uses 1448 bytes (0%) of program storage space. Maximum is 253952 bytes.
> Global variables use 9 bytes (0%) of dynamic memory, leaving 8183 bytes for local variables. Maximum is 8192 bytes.
> avrdude: ser_open(): can't open device "\\.\COM3": Access is denied.
> 
> 
> avrdude: ser_drain(): read error: The handle is invalid.
> 
> 
> avrdude: ser_send(): write error: sorry no info avail
> avrdude: stk500_send(): failed to send command to serial port
> avrdude: ser_recv(): read error: The handle is invalid.
> 
> 
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: ser_send(): write error: sorry no info avail
> avrdude: stk500_send(): failed to send command to serial port
> avrdude: ser_recv(): read error: The handle is invalid.
> 
> avrdude: ser_recv(): read error: The handle is invalid.
> 
> 
> 
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: ser_send(): write error: sorry no info avail
> avrdude: stk500_send(): failed to send command to serial port
> avrdude: ser_recv(): read error: The handle is invalid.
> 
> 
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: ser_send(): write error: sorry no info avail
> avrdude: stk500_send(): failed to send command to serial port
> avrdude: ser_recv(): read error: The handle is invalid.
> 
> 
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: ser_send(): write error: sorry no info avail
> avrdude: stk500_send(): failed to send command to serial port
> avrdude: ser_recv(): read error: The handle is invalid.
> 
> 
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: ser_send(): write error: sorry no info avail
> avrdude: stk500_send(): failed to send command to serial port
> avrdude: ser_recv(): read error: The handle is invalid.
> 
> 
> avrdude: stk500v2_ReceiveMessage(): timeout
> avrdude: stk500v2_getsync(): timeout communicating with programmer

As I said before, this is my first project with an Arduino ever, so if there is more information I can provide, I would be glad to provide it. I am having a heck of a time with this and I am sure it is because I am doing something wrong. 

Thank you for your help.
Dylan Fleming

dancind, 2019-11-01:
It would appear that the problem was with my computer. I have a Raspberry Pi handy, so I installed arduino on that and it works like a champ. Not sure what is going on with my computer.

per1234, 2019-11-05:
Thanks for the update @dancind. I'm glad to hear you managed to get it working. I'm going to go ahead and close this, but if you're still having problems please post on the Arduino Forum and we'll help you out over there.

closed 2019-11-05 by per1234