arduino/Arduino #9389: [Library Manager] New Library: EMUSerial

GTO2013, 2019-11-03:
Hi,
could you please add this repo to your list: https://github.com/GTO2013/EMUSerial

Thank you!

per1234, 2019-11-05:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

GTO2013, 2019-11-10:
Hi,
sorry for the late reply. I had already added a 1.0.1 tag to it, seems like something went wrong there. I created a release called 1.0.1 which matches the .property file.

cmaglie, 2019-11-11:
Done.

closed 2019-11-11 by cmaglie