arduino/Arduino #9403: Driver issue

snmnima, 2019-11-06:
I when upgraded all my device drivers, Arduino driver is not recognized by the computer.

PaulStoffregen, 2019-11-06:
How do you imagine anyone can understand and fix this problem with only this 1 simple sentence?  You didn't even say whether you're using Macintosh, Windows or Linux.  We don't know which Arduino board(s) you are using.  We have no idea what you actually did that is "upgraded all my device drivers", nor can see see what is or is not on your screen (or perhaps audible sound alerts) that you describe as "not recognized by the computer".

Usually Per will quickly mark these sorts of issues as invalid and close them.  I'm surprised I've seen it first, as he's almost always so fast & efficient.  Maybe you could add more info about what's really wrong before he comes along?  Remember, we can't see your screen or your hardware.  We're not psychic.  We can't know what things you are actually doing.  If you give us only your opinion and you omit the actual detail of what you really did and saw, odds are slim anyone can do anything with this issue.

pnndra, 2019-11-06:
Hi Paul,
thanks for addressing this. unfortunately Per is taking some time off for personal reasons so won't be very reactive in this period. at the same time the whole team is really under pressure for some Christmas presents we're preparing...
hope the community will understand our slow replies in these weeks.

PaulStoffregen, 2019-11-06:
Oh, I sure do understand not enough time for everything!

snmnima, 2019-11-08:
> How do you imagine anyone can understand and fix this problem with only this 1 simple sentence? You didn't even say whether you're using Macintosh, Windows or Linux. We don't know which Arduino [...]

Yes. you right. I'm sorry...
my Arduino board is Uno and I'm using windows 7 .
my Arduino board was fully identified by my computer but when I upgraded the all hardware by "Driver Pack Solution", it doesn't work right now!
I did this on another computer and had the same problem there and was using the "system restore" and fixed. I didn't create a "restore point" on my computer and the only way that comes to my mind is reinstall Windows!
thanks for suggesting me another way

per1234, 2019-11-15:
It sounds like this is a problem caused by a 3rd party program ("Driver Pack Solution") and thus is out of the scope of this issue tracker, which is for reporting bugs or feature requests with official Arduino software. So I'm going to close this.

Since I'm not familiar with this "Driver Pack Solution", it is possible that I have misunderstood your problem. If so, and this is actually related to a bug in official Arduino software, then please provide more details and I'll be happy to re-open. Most important would be to know which Arduino board (including the model of USB to TTL serial chip if the board does not have native USB) you have, which determines which driver it uses.

Regarding me not having been responsive to the issue trackers lately: I have been very ill the last few months to the point of not being able to work much recently. I'm starting to feel a lot better now and am hopefully moving back into fully operational mode. Thanks so much for your assistance with the issue reports Paul!

closed 2019-11-15 by per1234