arduino/Arduino #9406: [Library Manager] Add DCD SDK Arduino library in the library manager

livein21st, 2019-11-07:
Hi,

Please add my DCD SDK arduino library to library manager. The library allows student to connect to our IoT hub and send data.

Repository link:

https://github.com/datacentricdesign/dcd-sdk-arduino

cmaglie, 2019-11-07:
You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

livein21st, 2019-11-07:
I have added the release tag for git repo! Let me know if anything else missing.

Thank you for your support.

livein21st, 2019-11-08:
@cmaglie any updates?

cmaglie, 2019-11-11:
Done.

closed 2019-11-11 by cmaglie